PX4/PX4-Devguide #511: Quadcopter with asymmetrical/different arms length

GrEx0, 2018-04-16:
I am working on a quadcopter which has an asymmetrical configuration with different arms length (see picture below). Compared to a deadcat geometry my CoG is closer to the front arms.

[image]

where L1 = 746mm and L2 = 870mm.

To redefine the mixer matrix i referred to issue #435, but i think (correct me if i am wrong) that px_generates_mixes.py assumes constant arm length, which is not my case.

How can i redefine the mixer according to my situation?

Thanks for the support.

@jlecoeur

jlecoeur, 2018-04-16:
Hi @GrEx0 ,

The script does not assume constant arm length. So you can refine the mixer for your quadcopter! 

I suggest that you start from the deadcat geometry file. You will have to modify the "position" fields for the 4 rotors. The position field contains the X, Y, Z (front-right-down) position of the center of the rotor relative to the CoG. 
For example, the position of the front right rotor is `[L1 * sin(33), L1 * cos(33), 0]`.
And the position of the rear left rotor is `[- L2 * sin(44), - L2 * cos(44), 0]`

jlecoeur, 2018-04-16:
Yes it is in meters.
The unit does not matter as of today, but it might change later so it is better to use meters.

GrEx0, 2018-04-16:
Awesome, thank you :-)!