PX4/PX4-Devguide #518: Doc Bug: Preflight Checks · PX4 Developer Guide: Undocumented preflight failure

ndepal, 2018-04-25:
The following preflight failure is not documented:
"PREFLIGHT FAIL: EKF INTERNAL CHECKS"
It is printed [here](https://github.com/PX4/Firmware/blob/7bc3a12efe7f95f66e968d506829f1488e63480e/src/modules/commander/PreflightCheck.cpp#L539).

Bug Page: [Preflight Checks · PX4 Developer Guide](https://dev.px4.io/en/tutorials/pre_flight_checks.html)

hamishwillee, 2018-04-26:
Yes, already working on it in https://github.com/PX4/Devguide/pull/510

hamishwillee, 2018-04-26:
(and noting that this will move into the PX4 User Guide)

ndepal, 2018-04-26:
Sorry, didn't see that. That's great!

closed 2018-04-26 by ndepal

hamishwillee, 2018-04-27:
No worries - thanks for posting - for every issue I've already got ongoing work on I'm sure there are millions of things that I missed :-)

This got merged. It is now being moved into the USER guide.