PX4/PX4-Devguide #523: Unable to reproduce

Jaeyoung-Lim, 2018-05-11:
I am trying to do image streaming to QGC and was trying to reproduce the results from this [document](https://dev.px4.io/en/qgc/video_streaming.html)

However, when I run 
```
$ gst-launch-1.0 v4l2src device=/dev/video1 ! video/x-h264,width=1920,height=1080,framerate=24/1 ! h264parse ! rtph264pay ! udpsink host=127.0.0.1 port=5000
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2948): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.195888025
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```
My device on `/dev/video1` is a logitech webcam C170

I tried
```
$ gst-launch-1.0 v4l2src device=/dev/video1 ! udpsink host=127.0.0.1 port=5000
```

However, complains that the UDP packet is too large. 
```
$ gst-launch-1.0 v4l2src device=/dev/video1 ! udpsink host=127.0.0.1 port=5000Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
WARNING: from element /GstPipeline:pipeline0/GstUDPSink:udpsink0: Attempting to send a UDP packets larger than maximum size (614400 > 65507)
Additional debug info:gstmultiudpsink.c(715): gst_multiudpsink_send_messages (): /GstPipeline:pipeline0/GstUDPSink:udpsink0:

```

The v4l2src plugin itself seems to be working as 
```
gst-launch-1.0 v4l2src ! xvimagesink
```

is able to show the live image from the webcam.
I understand @hamishwillee is the author of the document. Any input in dealing with this problem will be appreciated.

Thank you in advance,

hamishwillee, 2018-05-14:
@kikislater Do you think you could advise on this - as the other author who has commented?

@Jaeyoung-Lim I did not write this article - I'm on the commit list because I subedited it/and possibly  copied it and updated it from elsewhere. I suggest you ask about this on the discussion boards or slack channel. Someone else may have tried it. If I had an Odroid I would try this, but I do not.

One alternative/recommendation you might try is to use the [Dronecode Camera Manager](https://hamishwillee.gitbooks.io/camera-manager/content/en/) instead of GStreamer directly. This basically handles the integration of your camera with PX4 on a linux platform, providing both RTSP streams and MAVLink Camera API endpoints for any connected camera. It still uses GStreamer, but the good thing is that it knows about what plugins work properly with QGC out of the box.

So either this will help you with the integration (it is how we're going to recommend integration with webcams/cameras in future) or it should at least help point you to where the likely sources of errors are.

kikislater, 2018-05-14:
Sorry I'm not an expert of gstreamer but I know there is a ton of change since 1.0 in comparison to previous 0.10
May be this issue could help : https://github.com/umlaeute/v4l2loopback/issues/97

Jaeyoung-Lim, 2018-05-14:
@hamishwillee  @kikislater  Thank you for the pointers,

It seems that it is an issue more related on the system side. I can stream images from the videotestsrc and ball patterns, but 
```
Internal data flow error
```
Shows on qgroundcontrol. 

When trying to use a ordinary gstreamer autovideosink, an issue comes with libva

```
gst-launch-1.0 udpsrc port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' ! rtph264depay ! avdec_h264 ! autovideosink fps-update-interval=1000 sync=false
Setting pipeline to PAUSED ...
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
Redistribute latency...
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
ERROR: from element /GstPipeline:pipeline0/GstUDPSrc:udpsrc0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2948): gst_base_src_loop (): /GstPipeline:pipeline0/GstUDPSrc:udpsrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:11.509275971
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

hamishwillee, 2018-05-15:
@Jaeyoung-Lim I know nothing about GStreamer either. I do know that QGC is only compatible with 
a small subset of the possible pipeline options:
```
muxer=rtph264pay
encoder=x264enc
converter=autovideoconvert
```
If that doesn't help, then I propose you raise this on the QGC slack channel, cross linking back to here.

lbegani, 2018-05-17:
@Jaeyoung-Lim I think the issue is with the gstreamer pipeline. Can you try with the following pipeline and see if it works
`gst-launch-1.0 v4l2src device=/dev/video1 ! video/x-raw,width=1920,height=1080! videoconvert ! x264enc !   rtph264pay ! udpsink host=127.0.0.1 port=5600`

Jaeyoung-Lim, 2018-05-17:
@hamishwillee  Where are the encoders and muxer defined in QGC?

hamishwillee, 2018-05-17:
No idea. That is a good question to ask, with context, in the QGC channel on Slack.

Jaeyoung-Lim, 2018-11-26:
@jinger26 Sorry for not coming back on this - I have solved this issue, it was a compatibility issue from the gstreamer version. I did not spend more time on identifying on which version works / breaks though

hamishwillee, 2018-11-28:
I'm going to propose this doc is removed, as per #666 
We should document gstreamer requirements of QGC, but this is not such a big deal if you use the camera manager, which already is configured for those requirements.

closed 2019-11-16 by Jaeyoung-Lim