PX4/PX4-Devguide #524: ULOG tools need quick description

LorenzMeier, 2018-05-11:
@bkueng We need one page in the dev guide about https://github.com/PX4/pyulog so people find them and can leverage them.

hamishwillee, 2018-05-14:
Pyulog is mentioned in a [bunch of places](https://dev.px4.io/en/?q=pyulog) in the devguide. People should be able to find them if they look up the obvious pages on Ulog:
* [Logging > Scripts](https://dev.px4.io/en/log/logging.html#scripts)
* [uLog File Format > Known Implementations](https://dev.px4.io/en/log/ulog_file_format.html#known-implementations)

Interestingly it is not mentioned in the topic [Flight Log Analysis](https://dev.px4.io/en/log/flight_log_analysis.html) and all the tools are buried in the above documents. 

Do you think the problem would be sorted if we moved all the [uLog File Format > Known Implementations](https://dev.px4.io/en/log/ulog_file_format.html#known-implementations) into [Flight Log Analysis](https://dev.px4.io/en/log/flight_log_analysis.html) instead? That is where I would expect to find the tools. Or do you think we really need an analysis tools topic?

bkueng, 2018-05-14:
I think we should keep the known implementations and extend https://dev.px4.io/en/log/flight_log_analysis.html to mention pyulog (with specific instructions how to e.g. convert to CSV). And while at it we can add https://github.com/Marxlp/pyFlightAnalysis as well (Roman is currently working on it)

hamishwillee, 2018-05-15:
@bkueng OK, left "known implementations" alone - makes sense - it covers things that aren't useful tools. 

I have taken your suggestion and added the two tools to the flight analysis page in https://github.com/PX4/Devguide/pull/525 

> (with specific instructions how to e.g. convert to CSV).

I've explained what tools are available and provided a single example of using one of them (not the convert tool). It is reasonable that we provide more detail on this tool since it is "owned" by PX4, but probably we can't do so in this document without making it unwieldy as an overview document. Some ideas:
- Create pyulog sub page with detailed instructions - reference that in this top level doc. 
- Add pyulog detailed instructions into readme for git repo.

What do you think?

On a related note, what is relevant to *Users*? Currently we have [Flying > Flight Reporting](https://docs.px4.io/en/flying/flight_reporting.html). IMO this could well be about the right level of info, but perhaps we should move to [Getting Started](https://docs.px4.io/en/getting_started/) to make it a lot more visible. 

Do you think the Flight Analysis and Flight Analysis Tools in Devguide is actually relevant to User Guide audience too? If so, that could be also moved as sub topics of Flight Reporting.

bkueng, 2018-05-15:
> I tend to think that the second option is better in this case (and I am happy to add such instructions)

I think so too. If you want to do that that would be great.

> On a related note, what is relevant to Users? Currently we have Flying > Flight Reporting. IMO this could well be about the right level of info, but perhaps we should move to Getting Started to make it a lot more visible.

I think it's good, and adding it to Getting Started makes sense.

> Do you think the Flight Analysis and Flight Analysis Tools in Devguide is actually relevant to User Guide audience too? If so, that could be also moved as sub topics of Flight Reporting.

This is where the line between users and devs blurs a bit, but I tend towards keeping it in the dev-guide only.

hamishwillee, 2018-05-17:
Great. All this now done as per linked items.

closed 2018-05-17 by hamishwillee