PX4/PX4-Devguide #528: Doc on creating the notification tunes (sounds)

hamishwillee, 2018-05-31:
Tunes are played on startup, error, and other notification. We need a doc explaining how this is done. It should cover:

- What are tunes
- What the default tunes (and are there non-default tunes?)
- Where are they defined? https://github.com/PX4/Firmware/tree/master/src/lib/tunes
- What is the tune format?
- How are they mapped to events?
- Who can create and use this service?
- What else would you like to know ....

This originates from https://github.com/PX4/Firmware/pull/9538

@MaEtUgR - would you be able to create first draft of this? Or bullet point the answers to the above and I'll create something!

MaEtUgR, 2018-06-11:
You should know that @simonegu wrote the new library that produces tunes.

* What are tunes
https://pixhawk.org/dev/tone_alarm
* Default Tunes
https://github.com/PX4/Firmware/blob/master/src/lib/tunes/default_tunes.cpp
Order according to https://github.com/PX4/Firmware/blob/master/src/lib/tunes/tune_definition.h
Non-default would be if a PX4 user or company using PX4 switches the melodies for certain events.
* Tune Format
is "ANSI Music" notation
http://artscene.textfiles.com/ansimusic/information/ansimtech.txt
* Who and how to play?
Currently tunes are ordered mostly by the commander module. With the "new" tunes library any part of the code could publish on the uORB topic `tune_control` https://github.com/PX4/Firmware/blob/master/msg/tune_control.msg to play something at any time. You can also play tunes from the PX4 command line via `tune_control` module, parameters: https://github.com/PX4/Firmware/blob/master/src/systemcmds/tune_control/tune_control.cpp#L66-L75 e.g. `tune_control play -m MNT150L64O3C<G>CEG>C<G<G+>CD+G+D+G+>CD+G+D+O2A+>DFA+FA+>DFA+F` Note that `tone_alarm` is now the driver only and doesn't support command line invocation to play sound directly anymore like discribed on https://pixhawk.org/dev/tone_alarm.

Does that help? 😇

hamishwillee, 2018-06-22:
Some additional info copied from pixhawk.org

====== Tone Alarm Driver ======

The tone alarm driver can play QBasic PLAY / ANSI music strings and drives a piezo buzzer depending on the frequency of the note.

  * [[http://en.wikibooks.org/wiki/QBasic/Appendix#PLAY|QBasic PLAY documentation]]
  * [[http://artscene.textfiles.com/ansimusic/|ANSI Music]]

The default tunes are stored in the [[https://github.com/PX4/Firmware/tree/master/src/drivers/stm32/tone_alarm|tone_alarm driver file]], but any tune can be played with a simple console command:

  tone_alarm MFT225O3L8GL8GL8GL2E-P8L8FL8FL8FMLL2DL2DMNP8

Note that in order to be recognised, a sequence passed to the tone_alarm command must begin with 'M'. ANSI music files typically also contain escape sequences at the beginning and end of the file that need to be removed before playing (as otherwise they tend to confuse the player).

The default meaning of the different tunes is:

^ Index     ^ Short name     ^ Description  ^
| 0         | STOP           | Stops the current tune  |
| 1         | BOOT           | 'PX4 startup sound'. Is auto-played by the tone alarm driver on system startup  |
| 2         | BATT_LOW       | The battery is low, but not fully empty. Running on reserve.  |
| 3         | BATT_EMPTY     | The battery is completely flat, the system is not able to maintain safe flight.  |
| 4         | CMD_ACK        | The last command has been accepted      |
| 5         | CMD_REJ        | The last command has been rejected      |
| 6         | POS_LOCK       | The position estimator obtained a valid position (emitted once on first lock)   |
| 7         | ARMING         | Emitted on the transition to armed   |
| 8         | DISARMING      | Emitted on the transition to disarmed   |
| 9         | EMERGENCY      | General, potentially fatal error. NOT due to battery.  Land immediately.   |

==== Creating new Alarm Tunes ====

ANSI music was popular in the days of ANSI BBS systems, and so the best editing tools are DOS utilities. One excellent option is [[ftp://archives.thebbs.org/ansi_utilities/melody21.zip|Melody Master]] combined with [[http://www.dosbox.com/|DosBox]].

  - Download Dosbox and install the app
  - Create a new directory in your home directory, e.g. ''dosprogs'', and unzip Melody Master to it
  - Mount the directory in dos with
<code dos>
mount c ~/dosprogs
</code>
  - Start Melody Master with:
<code dos>
  c:
  cd melody21
  start
</code>

{{ :dev:screen_shot_2013-03-10_at_11.49.59_am.png?500 |}}

==== Classics ====

  * [[http://www.ocarinaforum.com/index.php?topic=546.0|Tetris Theme]]

hamishwillee, 2018-06-22:
@MaEtUgR I copied the content from Pixhawk.org to above post. I can check what is relevant when we eventually create this doc, but FMI, is the Melody master any use for creating the music?

MaEtUgR, 2018-06-27:
@hamishwillee I used the melody master in dosbox so far. Didn't find any more modern application supporting the ANSI format output.

hamishwillee, 2019-02-04:
This is being addressed in https://github.com/PX4/px4_user_guide/issues/426

MaEtUgR, 2019-04-23:
The technical details are still not available. People constantly ask me for details... Can we at least reopen it again such that we don't forget.

hamishwillee, 2019-05-01:
@MaEtUgR What we have currently:
- Info on the [tune_control](https://dev.px4.io/master/en/middleware/modules_system.html#tunecontrol) module, which links to the source code [tune definitions and tune format](https://github.com/PX4/Firmware/blob/master/src/lib/tunes/tune_definition.desc).
- [Tune notification info for end users](https://docs.px4.io/master/en/getting_started/tunes.html) which also links to the definitions. 

What is it that you think is missing in terms of technical details?

MaEtUgR, 2019-05-01:
@hamishwillee Don't get me wrong you did a nice job with the existing docs you list above.

> What is it that you think is missing in terms of technical details?

Guide on how to compose, test and incorporate a custom tune because I think that's what a lot of developers will look for. Maybe I can make a linux and windows package and a quick rundown how to use it...

hamishwillee, 2019-05-02:
@MaEtUgR Makes sense, as this is non-obvious. 

Questions:
- To incorporate a custom tune, is it just a matter that you add a new `PX4_DEFINE_TUNE` at the end of the list in [tune_definition.desc](https://github.com/PX4/Firmware/blob/master/src/lib/tunes/tune_definition.desc#L108)?
- Can you define it out of tree? Ie so that you don't have to update it in PX4 source tree for new versions?
- What number should you use as a tune id? Can it just be the next free one?
- Are there any particular tricks to using a new tune once created?

With respect to creating/composing would it be reasonable to point them to https://github.com/PX4/Firmware/blob/master/src/lib/tunes/tune_definition.desc#L34 for the definition of the format. 

For testing/playing I tried a bunch of options, including melody master and qplay as per instructions above: https://github.com/PX4/Devguide/issues/528#issuecomment-399306391
Nothing appears to work on Windows.
What I did more recently is just attach hardware and play using `tune_control play -m "THE TUNE"`. Is this the best recommendation?

> aybe I can make a linux and windows package and a quick rundown how to use it...

A package for what?

MaEtUgR, 2019-05-03:
> To incorporate a custom tune, is it just a matter that you add a new PX4_DEFINE_TUNE at the end of the list in tune_definition.desc?

I'm not sure, so far I replaced existing tunes which a manufacturer might want to do for customization. I think it's possible to add a my_own_tune_definition.desc in the same folder and compile that one instead.

> What number should you use as a tune id? Can it just be the next free one?
> Are there any particular tricks to using a new tune once created?

I would need to check that. I was more referring to guide and tricks how to compose the tune but inserting and triggering it is also important, you're right.

> For testing/playing I tried a bunch of options, including melody master and qplay as per instructions above: #528 (comment)
Nothing appears to work on Windows.

Not sure what you tried but I always use this `melody21` DOS tool which works great on Windows (Dosbox). I wanted to put it in a simple to use package also for linux if that's possible.

hamishwillee, 2019-05-06:
Hi @MaEtUgR,

OK, managed to get melody21 working on windows. A package for Linux would be good. 

But when I write a tune and save it gets saved as a MEL. This does not look like our strings.  What is the trick to getting the save file in melody21 as a string we can put in our code?
```
 È   ü„ 1
 ” 2
œ 3
 ” 4
¬ 4
¬ 5
¬ 5
¬ 1
¬ 7
d   
```

So, our  OEM instructions in bullet form would be something like (ie what you want!):
1. Write your tunes according to the format defined at the top of Firmware/src/lib/tunes/tune_definition.desc
   - You can use a real device to test your tunes using a console OR you can (?) play them in melody21
   - I think we suggest that they can compose in melody21. Do you have any other specific tips and tricks to add?
2. To add your own sounds for existing notifications just modify the strings in tune_definition.desc for each tune (or maintain your own copy)
3. To add a new sound/tune you will need to define it and call it. To define it use PX4_DEFINE_TUNE with a new number - perhaps in a higher range that used by PX4 core? Or should we suggest they try add it into PX4 core and use next number? 
4. Then add some guidance on calling tunes.

I don't think anything else needs to be said?

MaEtUgR, 2019-05-07:
Somehow you can choose ANSI music output in melody21 and you'll get the string exactly like you need it for PX4.  I have to check. Yes, your outline is about what I would add to the documentation.
1. I didn't find a more modern program that allows this kind of editing and therefor I would suggest melody21 and make easy to use packages for windows and linux (mac I can't).
2. yes, instuctions on how to keep a customized version for OEMs.
3. That would be useful but I'd have to find out myself because I never did that before.
4. yes, testing, where they get used as a reference and how to call them in code.

hamishwillee, 2019-05-09:
Hi @MaEtUgR 

I worked out how to do ansi and documented above in https://github.com/PX4/Devguide/issues/528#issuecomment-399306391.

HOw do we get the useful string out of this:

[image]

When do you plan on looking at customised packages? I think we can get away without one for windows, but we need something for Linux.

Cheers

MaEtUgR, 2019-05-15:
Sorry for the delay. Yeah, that's exactly how it looks when you export from the ancient tool. The string from MNT to P64 is what you want... not optimal with all these escape characters around it. I'm not sure why they are there.

hamishwillee, 2019-05-16:
Thanks. Started this in https://github.com/PX4/Devguide/pull/792 for your review.

julianoes, 2019-10-09:
Have a look at the old docs: https://web.archive.org/web/20150326185600/http://pixhawk.org/dev/tone_alarm

hamishwillee, 2019-10-09:
@julianoes I captured everything in the old docs in https://github.com/PX4/Devguide/pull/792. Since @MaEtUgR didn't get around to reviewing I forgot about it. Merged now.

It does have one open question in it though, which is "how you go about adding a new tune (ie altogether). I'm pretty sure there is another issue somewhere for that.

PR #792 (2019-10-09): Tunes for developers docs
This fixes #528 . It is supposed to explain:
- [x] Tune architecture (lib, definitions, control)
- [x] Where default tunes are defined
- [x] How you can can create your own
- [x] How to replace existing
- [ ] How to add new in code.

@MaEtUgR FYI. This doesn't do the last one, because I don't have time to look into it yet. Also no non-Windows alternatives provided. 
Is it worth adding as-is and extending?
closed 2019-10-09 by hamishwillee