PX4/PX4-Devguide #547: [Outdated] Switching State Estimators · PX4 Developer Guide

bramsvs, 2018-06-21:
Page: [Switching State Estimators · PX4 Developer Guide](https://dev.px4.io/en/advanced/switching_state_estimators.html)

This page only tells half the story since you also need to set proper (LPE-enabled) build target as said here: https://dev.px4.io/en/ros/external_position_estimation.html, or include and start the modules manually.

hamishwillee, 2018-06-22:
Thanks @bramsvs 

@bkueng From [External Position Estimation](https://dev.px4.io/en/ros/external_position_estimation.html) it says that you have to build with the modules required present:

> Before following the instructions below, ensure that your autopilot has a firmware version with the LPE modules enabled. The LPE version of the PX4 firmware can be found inside the zip file of the latest PX4 release or it can be built from source using a build command such as make px4fmu-v2_lpe. See Building the Code for more details.

[Aside: Text "See Building the Code for more details." needs to be updated because it has no details on building LPE]

Is this correct? If you build with px4fmu-v2 will it only include EKA2? If you were to build with px4fmu-v2_lpe will it just include lpe or both lpe and ekf2? If not, how do you get both.

Also, https://dev.px4.io/en/advanced/switching_state_estimators.html only talks about MC and VTOL behaviour. How do you change things for FW?

bkueng, 2018-06-22:
>  If you build with px4fmu-v2 will it only include EKA2?

Yes, but it's only relevant for the v2 target. 

> If you were to build with px4fmu-v2_lpe will it just include lpe or both lpe and ekf2? If not, how do you get both.

Only LPE and you will not get both due to limited flash.

> How do you change things for FW?

LPE does not support FW.

In general:
- We should push for https://github.com/mavlink/qgroundcontrol/issues/6488 which fundamentally resolves this
- LPE is not maintained anymore and we should strongly advise everyone to use ekf2.

hamishwillee, 2018-06-22:
Thanks @bkueng , all that noted, but assume then I was after px4fmu-v5 and wanted both LPE and EKF2 - how then would I get both?

The reason I ask is not that I would recommend it, but that the rest of the docs talk about using parameters to specify the estimator used.

Or is it that you must build for either LPE or EKF2 and also set the parameters to make the estimator be started (ie you can't build them both in by default).

Cheers
H

bkueng, 2018-06-22:
> but assume then I was after px4fmu-v5 and wanted both LPE and EKF2 - how then would I get both?

v5 supports both already, no special build target is required - the same for every other except v2. By default (meaning on all non-v2 targets) you only need to change the parameter and reboot.

hamishwillee, 2018-06-25:
Fixed by https://github.com/PX4/Devguide/pull/549

bramsvs, 2018-06-27:
Nice :)

closed 2018-06-27 by bramsvs