PX4/PX4-Devguide #557: Hint to QGC console is missing in console guide

LorenzMeier, 2018-07-16:
@bkueng Could you do a quick pass?

https://dev.px4.io/en/debug/system_console.html

bkueng, 2018-07-16:
First needs this: https://github.com/mavlink/qgc-user-guide/issues/143

hamishwillee, 2018-07-16:
I will document that right now.

hamishwillee, 2018-07-17:
Note that this is somewhat mentioned here: https://dev.px4.io/en/debug/system_console.html#mavlink-shell

hamishwillee, 2018-07-18:
@bkueng Here you go https://docs.qgroundcontrol.com/en/analyze_view/mavlink_console.html

bkueng, 2018-07-19:
Thanks @hamishwillee. One note: the console can also be used via Telemetry, not only WiFi.

Devguide PR: https://github.com/PX4/Devguide/pull/560

hamishwillee, 2018-07-19:
@bkueng Hmm, the note was supposed to indicate that the benefit is deep access while flying, not that it is restricted to wifi. I have reworded to clarify: https://github.com/mavlink/qgc-user-guide/commit/1feb42d763c578549e8e3803fd561ac7723f3615

PR #560 (2018-07-20): system_console: add link to mavlink shell in QGC
Fixes  #557
closed 2018-07-20 by hamishwillee