PX4/PX4-Devguide #575: Doc Bug: UAVCAN Bus · PX4 Developer Guide

ndepal, 2018-08-08:
The list of CAN ESCs on this page is outdated.

According to @LorenzMeier (via slack):
>Orel 20 == Pixhawk ESC
The S2740 vanished

Bug Page: [UAVCAN Bus · PX4 Developer Guide](https://dev.px4.io/en/uavcan/)

hamishwillee, 2018-08-08:
@ndepal 

> "S2740 vanished" 

Do you mean this hardware is no longer available and this line should be deleted?

> Orel 20 == Pixhawk ESC

Orel 20 is an instance of the Pixhawk ESC design?

The Pixhawk ESC page is now invisible (ie the link went to pixhawk.org). However I checked it, and it links to [Sapog BLDC motor control solution](https://kb.zubax.com/display/MAINKB/Sapog+BLDC+motor+control+solution) zubax.com. This doesn't look very accurate - I THINK that perhaps this is ALSO an instance of Pixhawk.esc right?

Can you confirm that when we refer to PIXHawk ESC, we mean this reference design: https://github.com/PX4/Hardware/tree/master/sapog_reference_hardware

ndepal, 2018-08-09:
I don't know more than the quote from @LorenzMeier , so I can only guess.

> Do you mean this hardware is no longer available and this line should be deleted?

That's what it sounds like to me as well.

> I THINK that perhaps this is ALSO an instance of Pixhawk.esc right?

Well, the Orel 20 is the ESC designed by Zubax. It runs the Sapog firmware. And I believe it is based on the sapog_reference_hardware.

> Can you confirm that when we refer to PIXHawk ESC, we mean this reference design: https://github.com/PX4/Hardware/tree/master/sapog_reference_hardware

Can't confirm but sounds about right.

Ultimately it seems like the list of CAN ESCs should be shortened to just the Zubax Orel 20.
There is also the [Zubax Myxa](https://zubax.com/products/myxa) that has a CAN interface (among others) that might work, but I don't know.

hamishwillee, 2018-08-09:
Thanks @ndepal ! 

@LorenzMeier Anything to add? I "think" that once there was Pixhawk ESC reference hardware running Sapog firmware. Now there are various ESCs running sapog firmware on their own hardware. Is that correct - no pointers to Pixhawk ESC any more?

LorenzMeier, 2018-08-09:
Sapog == Pixhawk ESC. Codebase here: https://github.com/px4/sapog

LorenzMeier, 2018-08-09:
Reference HW: https://github.com/PX4/Hardware/tree/master/sapog_reference_hardware

hamishwillee, 2018-08-10:
Thanks. I *think* I have fixed in #578 - @ndepal I can't add you as a reviewer, but please check that link

closed 2018-08-10 by LorenzMeier

ndepal, 2018-08-10:
Sorry, I'm too late. But looks good to me.

One very minor thing is that it mentions "all motor controllers" below the list, but only one was introduced.
> All motor controllers provide status feedback and implement field-oriented-control (FOC).

hamishwillee, 2018-08-10:
That's a good point, but I am hoping we'll get some more to add in future. For now let's leave as is - not particularly harmful.

PR #578 (2018-08-10): Tidy supported ESC in UAVCan
Fixes #575