PX4/PX4-Devguide #585: Multi-Vehicle Simulation with jmavsim

hamishwillee, 2018-08-20:
Add the new multi-vehicle JMAVSim solution.
Possibly make this one a "parent": [Multi-Vehicle Simulation · PX4 Developer Guide](http://dev.px4.io/en/simulation/multi-vehicle-simulation.html).

The info reproduced from [here](https://github.com/Dronecode/DronecodeSDK/issues/494#issuecomment-413602890):

---

Multiple jMAVSim vehicles (with latest master):

Build PX4
```
make posix_sitl_default
```
Start multiple instances

```
Tools/sitl_multiple_run.sh 2
```

Then start the first instance:
```
 ./Tools/jmavsim_run.sh
```
Start subsequent instances, specifying the UDP port (expected to be 14560+i for i in [0, N-1]
```
./Tools/jmavsim_run.sh -p 14561
```

To have two drones connect on the same port, do this change:
```
diff --git a/ROMFS/px4fmu_common/init.d-posix/rcS b/ROMFS/px4fmu_common/init.d-posix/rcS
index 91c9cb7c8f..f276357d90 100644
--- a/ROMFS/px4fmu_common/init.d-posix/rcS
+++ b/ROMFS/px4fmu_common/init.d-posix/rcS
@@ -83,7 +83,7 @@ fi
 param set MAV_SYS_ID $((1+px4_instance))
 simulator_udp_port=$((14560+px4_instance))
 udp_offboard_port_local=$((14557+px4_instance))
-udp_offboard_port_remote=$((14540+px4_instance))
+udp_offboard_port_remote=$((14540))
 udp_gcs_port_local=$((14556+px4_instance))
 
 if [ $AUTOCNF == yes ]
```

hamishwillee, 2018-08-20:
@bkueng This is a new technique for me - we need to document it as it is much easier than alternatives like http://dev.px4.io/en/simulation/multi-vehicle-simulation.html

FYI, this froze for me in very latest master at `3D [dev] 1.6.0-pre12-daily-experimental daily` (the UI did start) in the Cygwin Windows environment

1. Does it work for Gazebo too (ie I can't see a Gazebo script to start instances)?
2. Any reason this was not documented? Ie is it not ready for that for some reason?

bkueng, 2018-08-20:
> @bkueng This is a new technique for me - we need to document it as it is much easier than alternatives like http://dev.px4.io/en/simulation/multi-vehicle-simulation.html

Exactly, it's the easiest way to run multiple vehicles. The drawback is that you also have multiple simulators.

> FYI, this froze for me in very latest master at 3D [dev] 1.6.0-pre12-daily-experimental daily (the UI did start) in the Cygwin Windows environment

Right, Cygwin Windows is broken on master, I think after https://github.com/PX4/Firmware/pull/10173.

> Does it work for Gazebo too (ie I can't see a Gazebo script to start instances)?

Gazebo should work too, but as you can see from the ROS setup, the vehicle model files (with the right parameters, such as position and ports) need to be generated. Should be doable with a script as well, but I have not looked into that. 

> Any reason this was not documented? Ie is it not ready for that for some reason?

No reason, it's just quite new.

hamishwillee, 2018-08-20:
Thank you. It is excellent. I hope to give it a go either tomorrow or in coming weeks and create a doc.