PX4/PX4-Devguide #612: octomap simulation

jbourne15, 2018-09-19:
I've been trying to run the simulation provided [here](https://dev.px4.io/en/simulation/gazebo_octomap.html) and I believe I have successfully installed everything.  However when I try launching the octomap_tracking_server.launch I get no occupied_cells_vis_array messages.

I have made changes in ~/catkin_ws/src/rotors_simulator/rotors_gazebo/CMakeLists.txt  and package.xml.

I have also made changes in octomap_tracking_server.launch which relabel the fixed frame to world and the pointcloud2 cloud_in to be /firefly/vi_sensor/camera_depth/depth/points.

By opening rviz I can see that I have a pointcloud2, but no occupied_cells_vis_array. I have the following warning from octomap_talker:  Nothing to publish, octree is empty.  This prints out once and then proceed on by printing: starting server. 

I am on kinetic, ubuntu 16.04. Is this tutorial up to date for kinetic?

I am not sure what I am doing wrong, but any help would be great

mhkabir, 2018-09-19:
That tutorial is hopelessly out of date and should be removed.

jbourne15, 2018-09-19:
Ok, I still think this should work but maybe this is a question for a different place. thanks anyways.

hamishwillee, 2018-09-19:
@jbourne15 

As @mhkabir says, the tutorial is probably very out of date. I'm not quite so ready to delete - my take is "if you can work out how to fix it we'll be very happy to take a PR".

@TSC21 have you tried this/got any advice? Is the tutorial recoverable? (if not, then deletion is probably a good idea).

TSC21, 2018-09-19:
@hamishwillee yes they are recoverable. It's a matter of sync up things with the current `sitl_gazebo` available models.

jbourne15, 2018-09-19:
@TSC21 @hamishwillee I got things working with iris_rplidar model. Also I had to convert the laser/scan to a pointcloud2, remap this topic and change the fix frame and use (octomap_server) octomap_mapping.launch, not octomap_tracking_server.launch.

TSC21, 2018-09-19:
@jbourne15 great! Can you please issue a PR to the devguide with the suggested changes you did? Thank you!

hamishwillee, 2018-09-20:
> @jbourne15 great! Can you please issue a PR to the devguide with the suggested changes you did? Thank you!

Yes please @jbourne15 ! 

There are instructions on how to do docs updates [here](https://dev.px4.io/en/contribute/docs.html). The files are just markdown. As long as you don't want to modify images or create new files you can probably do a "quick update" (ie all the changes in github editor).

jbourne15, 2018-09-22:
I'll see what I can do, it might be a little bit before I can spend sometime on this.

hamishwillee, 2018-09-23:
Thanks @jbourne15 - that would be much appreciated. Even pointing to your code branch or jotting some notes down might be useful.