PX4/PX4-Devguide #638: MPC_HOLD_MAX_Z MPC_HOLD_MAX_XY description

tuloski, 2018-10-25:
It is not clear what the above parameters do.
From the description: Maximum horizontal velocity for which position hold is enabled.

But what is position hold? Is the flight mode "Position"? What happen when the velocity is higher than the parameter? It switches to manual?

hamishwillee, 2018-10-26:
> But what is position hold? Is the flight mode "Position"? 

Yes.

My *guess* below - @bkueng , can you confirm?:

When you're in position mode PX4 attempts to hold your current position against wind and other external forces. When you deliberately start to move somewhere you apply thrust. This will overcome the effects of position hold and you will start moving. Once you reach the velocity in these parameter setpoints position hold will turn off. 

Beat, if that is correct, 
- does it mean that if a strong enough wind came to move you at these velocities position hold would turn off and give up? I guess that would make sense because the motors would already have failed to hold you in place.
- Why doesn't position hold just turn off when it detects a certain velocity setpoint?

bkueng, 2018-10-26:
> does it mean that if a strong enough wind came to move you at these velocities position hold would turn off and give up?

No, it will continue to try to hold it.

> Why doesn't position hold just turn off when it detects a certain velocity setpoint?

It does.

There's some explanation for this here: https://docs.px4.io/en/config_mc/advanced_mc_position_tuning.html
Basically it operates in 2 modes: position hold when your sticks are centered AND (horizontal) velocity is below `MPC_HOLD_MAX_XY`. Which means the vehicle will keep its position and not drift. If the condition is not met, the vehicle is velocity-controlled, which also means it can drift (which is usually ok since you want the vehicle to move).
Is this clear?

tuloski, 2018-10-26:
Ok thanks for the explanation. So basically it turns off the P of the position loop, but keeps the PID on the velocity. It's a bit weird from control theory point of view, but I understand the practical case.
What I don't get is if it turns off position when the velocity is big enough of when the velocity setpoint is big enough.
Which are the drawbacks or advantages to put the parameter at 0, i.e. disable this feature?

hamishwillee, 2018-10-28:
> Is this clear?

Possibly - let me confirm. What I think you are saying is that this parameter defines when position control turns on. So you're speeding along in velocity control and you release the sticks. Velocity control starts to force the velocity to zero. When the speed drops below [MPC_HOLD_MAX_XY](https://docs.px4.io/en/advanced_config/parameter_reference.html#MPC_HOLD_MAX_XY) the vehicle will start position control. The vehicle is latch in position control until the sticks move out of the deadzone.
- By contrast if you're in position control and you move the sticks out of the deadzone the vehicle immediately moves to velocity control.
- Once you're in position control mode with sticks in deadzone the mode is latched - so even if the wind causes the vehicle to move at greater velocity position mode will remain on.

Is that correct?  (@tuloski FYI - @bkueng is on holiday this week, so I am not expecting an answer until next week!)

> Which are the drawbacks or advantages to put the parameter at 0, i.e. disable this feature?

So if I am correct above, a low number will increase the drift time before you lock to a position. On a windy day, in theory if moving with the wind you might not lock anywhere near where you released the sticks. By contrast, if the value is high you'd probably have significant overshoot of your target point. 

> How do we capture this discussion?

My assumption is that generally the default is good enough, which is why this isn't captured in the tuning docs. I think it is best that we do capture it there, if only so that this can be found on search. 

Assuming my assumption is correct, how about add section to https://docs.px4.io/en/config_mc/advanced_mc_position_tuning.html#position_mode like this:

## MPC_HOLD_MAX_XY / MPC_HOLD_MAX_Z

The [MPC_HOLD_MAX_XY](https://docs.px4.io/en/advanced_config/parameter_reference.html#MPC_HOLD_MAX_XY) and [MPC_HOLD_MAX_Z](https://docs.px4.io/en/advanced_config/parameter_reference.html#MPC_HOLD_MAX_Z) parameters define the velocity at which a moving vehicle will change from velocity control mode to position control mode when the sticks are released/centred. 

If a low velocity is specified the vehicle will travel further before locking to a position, while a high value may result in the vehicle "overshooting" and then returning to the locked position. 
The default value is appropriate for most vehicles.

> **Note** Once in position control mode the vehicle will remain in the mode even if external forces result in a velocity greater than the parameters. 
  The vehicle returns to velocity control mode only when the sticks are moved out of the centre zone.

bkueng, 2018-11-05:
@tuloski you find the exact logic here: https://github.com/PX4/Firmware/blob/master/src/lib/FlightTasks/tasks/ManualPosition/FlightTaskManualPosition.cpp#L119. So it's the velocity.
There's also a new algorithm in development, see https://github.com/PX4/Firmware/pull/10696. You're welcome to test & contribute there.

@hamishwillee you are correct. You I'd wait with updating the docs, as there is further development going on.

bkueng, 2018-11-06:
> In any case, can you ping me when you think docs need new update?

Yes.

bresch, 2018-11-07:
Note: in the new position control based on limited jerk trajectories, the position lock/unlock is automatically done when the velocity and acceleration of the trajectory is below a certain value. The position hold "kick" is not noticeable anymore and this parameter is not used.
To enable this new trajectory setpoints generator, set the `MPC_POS_MODE` to 3 (check [here](https://github.com/PX4/Firmware/pull/10696) for the tuning parameters).