PX4/PX4-Devguide #639: FW_LND_USETER

zanemountcastle, 2018-10-26:
It is unclear what exactly this parameter does. The parameter reference just says:

> Use terrain estimate during landing

although the source of this terrain estimate isn't clear. 

Is the terrain estimate different from the fused altitude? If a range sensor is enabled, will it default to that? Will it use the terrain altitude that QGroundControl now uses for mission planning?

This was referenced by [this issue](https://github.com/PX4/Firmware/issues/7289) on the Firmware repository, but nothing was ever really clarified before the issue went stale.

hamishwillee, 2018-10-28:
@zanemountcastle This uses ground altitude from GPS instead of the typical return altitude or waypoint altitude. See https://docs.px4.io/en/flying/fixed_wing_landing.html#FW_LND_USETER

zanemountcastle, 2018-10-28:
@zanemountcastle Thanks for the link. Maybe the detail from that page should also be included in the full parameter reference, but nevertheless it's good to see that it exists.

On another related note, how exactly would a landing work with the `FW_LND_USETER` parameter enabled? Would the plane still follow the `FW_LND_ANG` until it detects that it has landed or would the plan attempt to level out with the terrain as it expects to be the ground from the terrain estimate? 

Note: In a recent flight, I enabled this parameter and the plane leveled out at about 30m on its final approach until I switched into `Stabilized` mode and landed it manually. At the location where I was trying to land, Google Maps GPS altitude says the altitude is 167.5m whereas the altitude recorded at takeoff was about 155m. For the full flight logs see here: https://logs.px4.io/plot_app?log=e4d003b1-cee4-4c13-9069-34e5e04ebcf5

I'm still having trouble diagnosing exactly what went wrong. 

[image]

zanemountcastle, 2018-10-28:
@hamishwillee Thanks for the link. Maybe the detail from that page should also be included in the full parameter reference, but nevertheless it's good to see that it exists.

On another related note, how exactly would a landing work with the `FW_LND_USETER` parameter enabled? Would the plane still follow the `FW_LND_ANG` until it detects that it has landed or would the plan attempt to level out with the terrain as it expects to be the ground from the terrain estimate? 

Note: In a recent flight, I enabled this parameter and the plane leveled out at about 30m on its final approach until I switched into `Stabilized` mode and landed it manually. At the location where I was trying to land, Google Maps GPS altitude says the altitude is 167.5m whereas the altitude recorded at takeoff was about 155m. For the full flight logs see here: https://logs.px4.io/plot_app?log=e4d003b1-cee4-4c13-9069-34e5e04ebcf5

I'm still having trouble diagnosing exactly what went wrong. 

[image]

hamishwillee, 2018-10-29:
Hi @zanemountcastle 

> Maybe the detail from that page should also be included in the full parameter reference, but nevertheless it's good to see that it exists.

Good idea: https://github.com/PX4/Firmware/pull/10777
This is open source - so feel free to update this sort of thing yourself.

> On another related note, how exactly would a landing work with the FW_LND_USETER parameter enabled? 

So *my understanding* is that all this does is set where the vehicle thinks the ground is. When this isn't set the vehicle might use home location  on return or the mission altitude. If this is set you're using GPS altitude - which is probably the WGS84 ellipsoid guess at earth surface. That would explain why the altitude doesn't match the reality. 

> until it detects that it has landed or would the plane attempt to level out with what it expects to be the ground from the terrain estimate?

That I do not _know_ (note, I would call it "altitude datum" rather than "terrain estimate"). I would _expect_ it to continue to follow the glide slope/flare path and very slowly drop (the horizontal scale is compressed in the landing diagram, so the actual landing path is much less steep in reality).

@dagar can you confirm this last point? (if the altitude datum is well above the actual ground, will the vehicle effectively level out, very slowly decreasing in altitude as it follows the flare path?)

dagar, 2018-10-29:
The terrain estimate in this context is coming from use of a distance sensor (usually a laser range finder). This includes flying the calculated approach based on landing angle.

zanemountcastle, 2018-10-29:
@dagar The plane here had a distance sensor enabled and working properly, but it still remained level at about 30m where it was supposed to be landing. In other words, it seemed to ignore the landing angle despite being on its final approach--I'm guessing because it estimated the ground altitude to be higher than the real ground altitude because of the `FW_LND_USETER` parameter and inaccurate global terrain altitude data. Maybe I'm just not understanding how the plane calculates its relative altitude, but it seems to me like the `FW_LND_USETER` parameter overrode the conventional descent at a rate determined by `FW_LND_ANG` until it detects landing.

I dug into the source code a little (see below) and it looks like it won't use the local terrain estimate (I'm guessing this is determined by the distance sensor) and instead will continue with the global terrain altitude data (assuming this data is determined to be "valid"). The issue seems to have arisen where the autopilot recognized the global terrain data as "valid," even though it seems to be off by at least 15m or so. Maybe there should be a way to locally override the terrain data if it identifies a discrepancy between where the ground _should be_ (based on global terrain data) and where the ground _is_ as determined by a distance sensor. 

https://github.com/PX4/Firmware/blob/a7d297ed574a9b1110a224eb3718fbe14b154f6d/src/modules/fw_pos_control_l1/FixedwingPositionControl.cpp#L1525-L1557

zanemountcastle, 2018-11-01:
@dagar @hamishwillee thoughts?

hamishwillee, 2018-11-01:
@zanemountcastle I have no thoughs :-). @dagar is away today/tomorrow. ONce he has commented my interest is in making sure that the docs state unambigously what is/should happen here. 

Please do continue to ping until this is sorted!

zanemountcastle, 2018-11-01:
@hamishwillee Sounds good.

Another thing to note is that I ran a test flight yesterday with everything the same except I disabled the `FW_LND_USETER` parameter and the plane landed wonderfully. It seems like in fact it _was_ this parameter that was causing the plane to overfly the runway way.

The flight logs are here for reference: https://logs.px4.io/plot_app?log=bd17fe0a-8d99-4cf4-b1ce-f6ab7fc48b8f

zanemountcastle, 2018-11-06:
@dagar Did you get a chance to look into this?