PX4/PX4-Devguide #644: Parameters & Configs  - Not enough to create my own

hamishwillee, 2018-11-12:
[Parameters & Configs](https://dev.px4.io/en/advanced/parameters_and_configurations.html) is supposed to explain how to work with parameters. IMO it is not enough to do so.

Specifically we need 
* [Parameter Metadata](https://dev.px4.io/en/advanced/parameters_and_configurations.html#parameter-meta-data)
  * What file should these be declared in - where should it be located, file extensions or naming conventions etc, should it be per module or per library or what?
  * What are naming conventions for individual parameters?
  * What types of params are supported - e.g. `PARAM_DEFINE_FLOAT`. Where is API for this?
  * HOw are the params imported into a module/modules?
* How can their values be accessed and set from C++ code. 

I also looked at the [Module Template](https://dev.px4.io/en/apps/module_template.html) which states that it shows "Accessing parameters and reacting to parameter updates.". HOwever I can't see the parameters [being defined with metadata](https://github.com/PX4/Firmware/blob/master/src/templates/module/module.h#L77).

I would consider this sufficiently documented if I could have just sent this page link as a response to this comment: 
https://github.com/PX4/Firmware/issues/10822#issuecomment-437673911

@bkueng Can you help me by answering some of the above? (or editing the page appropriately)?

hamishwillee, 2018-11-12:
FYI, just checking and this duplicates https://github.com/PX4/Devguide/issues/532 . I looked at that, and it does not answer all of the above questions.

bkueng, 2018-11-12:
- The parameter metadata is in a separate file, typically within the module. For example https://github.com/PX4/Firmware/blob/master/src/drivers/camera_trigger/camera_trigger_params.c. There's only `PX4_PARAM_DEFINE_FLOAT` and `PX4_PARAM_DEFINE_INT32`.
- Then you have 2 options to use it:
  - The (older) C API: https://dev.px4.io/en/advanced/parameters_and_configurations.html#c--c-api
  - Use the newer C++ API, which the Module Template is using.

That should be all. Do you need more info?

hamishwillee, 2018-11-14:
@bkueng Ha ha, not even close.

I've implemented triggers using old style API in https://github.com/hamishwillee/Firmware/pull/1 - can you please answer the questions there?

For the new API, I can't really tell what is essential and what is fudge. 
Can you confirm it works something like:
1. Modules include `#include <px4_module_params.h>` to (ultimately) get the required DEFINE_PARAMETERS macro. 
   - Should the header be included into drivers as well, or is it just for modules?
   - Can the DEFINE_PARAMETERS be used anywhere or are there restrictions?
2. The params in module are declared like this:
   ```
   	DEFINE_PARAMETERS(
		(ParamInt<px4::params::SYS_AUTOSTART>) _sys_autostart,   /**< example parameter */
		(ParamInt<px4::params::SYS_AUTOCONFIG>) _sys_autoconfig  /**< another parameter */
	)
   ```
   - I presume the only requirement here is that `SYS_AUTOSTART` matches your docs/whatever the param is being sent
   - I presume from now on in your code you just use _sys_autostart and _sys_autoconfig  ?
   - I am guessing that the rest of the code is boilerplate to automatically update _sys_autostart  and _sys_autoconfig if the underlying parameters change. So other than this pointerplate you don't need to do anything. Is that correct?
   
 3. So for the update, I believe this is magic via uORB. It is hard to be sure because https://dev.px4.io/en/middleware/uorb.html is not at all useful. There isn't much about the API - for orb_subscribe and orb_check and orb_copy are mentioned but not documented. [Can you give that a scan and confirm whether I'm correct - this is not very good starting point]. Anyway, on guess:
    - `#include <uORB/topics/parameter_update.h>` - allows you to subscribe to parameter updates via uORB. This topic will be a map to incoming PARAM_SET MAVLink messages.
   - We subscribe to parameter updates at beginning and unsubscribe at the end. `parameter_update_sub` returned by `orb_subscribe` is a handle we can use to refer to this particular subscription?
      ```
      int parameter_update_sub = orb_subscribe(ORB_ID(parameter_update));
      ...
      orb_unsubscribe(parameter_update_sub);
      ```
   - We call `parameters_update(parameter_update_sub, true);` periodically to check if there has been an update.
      - orb_check just tells us if there is any update to the param update message and puts it in a `parameter_update_sub`
      - We then do `orb-copy` to copy the updated value from parameter_update_sub into a local struct `param_upd`. 
       - Last of all, we call `updateParams();` is this where the magic happens that updates the SPECIFIC values in our DEFINE_PARAMETERS list with the right updates?

```
void Module::parameters_update(int parameter_update_sub, bool force)
{
	bool updated;
	struct parameter_update_s param_upd;

	orb_check(parameter_update_sub, &updated);

	if (updated) {
		orb_copy(ORB_ID(parameter_update), parameter_update_sub, &param_upd);
	}

	if (force || updated) {
		updateParams();
	}
}
```

bkueng, 2018-11-15:
> Should the header be included into drivers as well, or is it just for modules?

Wherever you want to use the `DEFINE_PARAMETERS`. So in drivers too.

> Can the DEFINE_PARAMETERS be used anywhere or are there restrictions?

This is meant to be used only for class attributes.

> I presume the only requirement here is that SYS_AUTOSTART matches your docs/whatever the param is being sent

Right

> I presume from now on in your code you just use _sys_autostart and _sys_autoconfig ?

yes

> I am guessing that the rest of the code is boilerplate to automatically update _sys_autostart and _sys_autoconfig if the underlying parameters change. So other than this pointerplate you don't need to do anything. Is that correct?

Correct

The uORB API doc is here and linked: https://github.com/PX4/Firmware/blob/master/src/modules/uORB/uORBManager.hpp
The rest is all correct.

> Last of all, we call updateParams(); is this where the magic happens that updates the SPECIFIC values in our DEFINE_PARAMETERS list with the right updates?

Correct. Lots of magic :D

PR #657 (2018-11-22): Update parameter doc with API and metadata info
This adds improved parameter docs, including info on the C and C++ APIs and the param metadata. 
Fixes #532
Fixes #644

@bkueng This is a good first draft - meaning I think it is correct, but I'm pretty tired so I will read it again tomorrow. It is worth your while checking for accuracy and omission.
closed 2018-11-22 by bkueng