PX4/PX4-Devguide #650: MAVLink Messaging - update to MAVLink 2

hamishwillee, 2018-11-14:
[MAVLink Messaging · PX4 Developer Guide](http://dev.px4.io/en/middleware/mavlink.html) needs update work.

From: https://github.com/PX4/Devguide/pull/649#issue-230708462

1. It is based on MAVLink 1.0 and needs updates to 2.0? I THINK that is just a matter of changing the links. 
2. This starts at the point you already have a library with the new message available. However it should start a bit earlier. Specifically, what is the process for me to add my own new generated library into a build? Is it just a matter of building the library and dumping it into Firmware/mavlink/include/mavlink/ ?
   - My reasoning is that if the message is custom, you aren't getting this into the standard library. It isn't necessarily obvious the "right" way to get the message into PX4 for testing.

@bkueng I'm assigning to you, but you can push back to me with a few notes on how this would be done if you like!

hamishwillee, 2018-11-15:
Also https://github.com/PX4/Devguide/issues/538

bkueng, 2018-11-16:
> It is based on MAVLink 1.0 and needs updates to 2.0? I THINK that is just a matter of changing the links.

Correct. And the generator needs to be configured to output mavlink 2.0.

> Specifically, what is the process for me to add my own new generated library into a build? Is it just a matter of building the library and dumping it into Firmware/mavlink/include/mavlink/ 

Correct, once you have generated the lib, just copy everything over to `Firmware/mavlink/include/mavlink/v2.0`.

>  My reasoning is that if the message is custom, you aren't getting this into the standard library. It isn't necessarily obvious the "right" way to get the message into PX4 for testing.

Making it available to others to test is a bit of a hassle. You would do:
- fork https://github.com/mavlink/c_library_v2 (for simplicity, it's just as valid to create an entirely new repo)
- add your generated lib there and push the changes
- let others know to update the submodule to the changed fork.

> @bkueng I'm assigning to you, but you can push back to me with a few notes on how this would be done if you like!

Going for the notes then.
Is it clear?

bkueng, 2018-11-19:
> (I assume it is just go into the submodule directory, use git as usual to set remotes, fetch branch on remote, checkout branch on remote)

Yes