PX4/PX4-Devguide #651: Doc Bug: Flight Controller Porting Guide · PX4 Developer Guide

mrpollo, 2018-11-15:
The guide doesn't cover the steps for upstreaming a new port, regardless of type (NuttX/Linux)

Questions:
* at what point is it recommended to reach out to the project maintainers?
* How does a review work, what are the requirements for such pull request?
  * Where can I ship my hardware for review?
  * What about flight testing? should we send hardware for testing as well?
* How should I handle inconsistencies in the FMU implementation of my new board?
* What level of maintenance and support should I expect from the PX4 project?
  * What is the recommended role I should take as a manufacturer/maintainer of a port?

Bug Page: [Flight Controller Porting Guide · PX4 Developer Guide](https://dev.px4.io/en/hardware/porting_guide.html)

hamishwillee, 2018-11-15:
https://github.com/PX4/Devguide/issues/368

hamishwillee, 2018-11-15:
@mrpollo Do you know the answer to any of these questions? Just to check terminology, by "upstreaming" you mean for someone with a new flight controller board to add it to PX4 codelines?

hamishwillee, 2018-11-15:
@LorenzMeier @dagar @davids5 - can you expand on these?

mrpollo, 2018-11-16:
Someone on forums wrote instructions
http://discuss.px4.io/t/how-to-build-for-a-new-flight-controller-hardware-target/8497

mrpollo, 2018-11-16:
I will try to answer the questions based on research I'm doing right now, I'll post back what I find.

"Upstreaming" I meant pushing code to our repo.

@all, I know we don't have answers to some of the questions above, the idea is to start a discussion and document as much as we can, this post is the first step.

LorenzMeier, 2018-11-16:
@mrpollo At the end of the day there is a very simple rule: Someone needs to sponsor a board, which means recurring maintenance cost, including hardware testing. If the contributor is not credibly capable of that there is nothing they can do to convince me to add it.

That is *in contrast* to software features that benefit everybody. For those we assume the maintenance resposibility. For board support only one vendor benefits.
PX4 is the "vendor" of the FMU series reference boards and member companies like Auterion provide the manpower for it.

* at what point is it recommended to reach out to the project maintainers?

When you are ready to contribute *and* maintain the contribution.

* How does a review work, what are the requirements for such pull request?

It needs to pass the full test suite and flight testing. To be eligible, the product needs to be available in the market.

  * Where can I ship my hardware for review?

Contact lorenz@px4.io for guidance.

  * What about flight testing? should we send hardware for testing as well?

Yes, that is mandatory.

* How should I handle inconsistencies in the FMU implementation of my new board?

Depending on the severity of the deviations the board might not be eligible for upstream support.

* What level of maintenance and support should I expect from the PX4 project?

PX4 only supports the FMU standard and the software. There is no maintenance and support for deviating builds.

  * What is the recommended role I should take as a manufacturer/maintainer of a port?

If the board is accepted upstream and not part of the FMU standard, the sole responsibility for maintaining it lies with the manufacturer. Failure to maintain the board will lead to its removal from upstream support.

hamishwillee, 2018-11-19:
I have attempted to capture this, and also explain why you might want to upstream in https://github.com/PX4/Devguide/pull/654