PX4/PX4-Devguide #656: Add docs for posix dynamic modules on Posix + FIFOs

hamishwillee, 2018-11-20:
Support for dynamic modules on Posix (and possibly nuttx, untested) added in https://github.com/PX4/Firmware/pull/10870#issuecomment-439932956

Needs to state what it can be used for and how it is used.
Possibly add a section in  [System Startup](http://dev.px4.io/en/concept/system_startup.html) linked from a small note in the [app tutorial](http://dev.px4.io/en/apps/hello_sky.html).

In addition, the section in [System Startup](http://dev.px4.io/en/concept/system_startup.html) about client/daemon FIFOs is no longer true. 

Hoping that @m-ou-se will create PR :-)

m-ou-se, 2018-11-23:
Updated documentation on the client/daemon communication in #662.

Will submit a PR for the dynamic modules as soon as https://github.com/PX4/Firmware/pull/10870 is merged.

m-ou-se, 2018-11-24:
The code is merged. #663 updates the documentation.

PR #663 (2019-05-30): Mention loadable modules in system startup and the app tutorial.
Fixes #656
closed 2019-05-30 by hamishwillee