PX4/PX4-Devguide #668: uorb top lost messages not clear

tuloski, 2018-11-26:
The documentation says (here: https://dev.px4.io/en/middleware/uorb.html) that the 5th column is the number of lost messages, but it is not clear if it is lost messages per seconds, lost messages total...
From my test the number can go up and down, so for sure it is not the total message lost. Nor can be the messages lost per second because it's higher than the frequency of the topic. Maybe it's message lost per second multiplied by the number of subscribers?

bkueng, 2018-11-27:
It's lost per second, but the sum for all subscribers.

hamishwillee, 2018-11-28:
FYI, clarified in https://github.com/PX4/Devguide/commit/dc7efaa9c5d3291110b44fde334416f74bdbd9c3

closed 2018-11-28 by hamishwillee