PX4/PX4-Devguide #676: New infrastructure to support versioned docs builds (Dev/User guide)

hamishwillee, 2018-11-28:
Dev and User guide reflect state of "master" but many users are working on the latest stable version (or older versions). This results in confusion when new features or changes are added to the docs. The problems increase as the master branch diverges. 

This is difficult to manage by tagging up changes in master - we don't always know what releases things will go into, and we don't want to have both new and old instructions in the docs - that quickly moulders. 

The best solution is to spawn off a separate doc set for each new version. We have a plugin already in place to support this, but we will need to update our Jenkins and URLs to enable it.

There are two options for spawning off the doc set:
- Create release version docs based on tags. Docs build once for each tag. There is no concept that we maintain old docs, and any mistakes cannot be corrected. 
- Create docs based on branches. Docs are built for release branches and are rebuilt/updated as changes are pushed to those branches.

@dagar I would like docs based on branches because it provides the greatest flexibility. But I don't know what is harder or easier to implement - either would "do" - so just let me know. 

The final result (assuming branch used for release docs)
- [ ] Docs for Devguide master branch accessible at https://dev.px4.io/master/* . 
  - Rebuilt just as currently based on changes to the devguide master branch.
- [ ]  Redirect current URLs:
  - [ ] https://dev.px4.io/ TO https://dev.px4.io/master/ (ie root URL)
  - [ ] https://dev.px4.io/en/* TO https://dev.px4.io/master/en/*
    - All other paths unchanged - no need to redirect the /zh, /kr paths.

**Either/or** the following (first preferred):
- [ ] Docs for branches built to `https://dev.px4.io/<branch_name>/`
  - Docs for branches built when branch created, updated with branch updates, and deleted with branch deletes.
- [ ] Docs for tags built to `https://dev.px4.io/<tag_name>/`
  - Docs for tags built when tag created. Docs for tags removed if tag deleted. No updates required/possible.

 - [ ] Docs for User Guide exactly the same pattern as above.

bys1123, 2018-12-28:
[image]
I'm developing a website compatible with gitbook, make sure chinese developer can switch versions at mirror site in china, and make translation update easier.
https://github.com/DimianZhan/ThiOne
It's based on Laravel, a very powerful php framework, getting popular in recent years, easy to add new features. It's not fully functional yet, will be ready in two weeks.

hamishwillee, 2018-12-31:
@bys1123 That looks pretty nice. Gitbook actually has a solution for versions which you can see in action here: https://sdk.dronecode.org/en/
We plan to turn this on in the coming days (ie close this issue). 

It is good to know that there are other compatible solutions out there for markdown. I wouldn't probably switch to another system without a compelling reason (which version support is not, given that we just need to enable it). I'm also a big fan of truly static sites because it makes hosting easier and cheaper to setup and maintain than one which requires a programming environment. 

A compelling reason to update to a system with some sort of backend would be a fantastic collaborative editing experience. Gitbook are trying to achieve this but are still some way from achieving it.

bys1123, 2018-12-31:
@hamishwillee Understand. I'll also try to make it better, looks hard, but programming is fun.

The reason I did not still use gitbook, because of the version plugin only support the "branch" method when you use gitbook.com. If use manual configuration, experience is not that good.

https://plugins.gitbook.com/plugin/versions

hamishwillee, 2018-12-31:
Hi @bys1123 
The branch method is actually better for docs (usually) because there is often a need to go back and tidy mismatches in the old docs. For example, if I take a snapshot of docs right now it includes some stuff that will not be in the next release. 

> That will need edit the books.json in each version, also static pages is in _book folder in each version, need to config the url rewrite in Nginx or Apache.

Incorrect. The selector for all builds is picked up dynamically from what you put in `gitbookConfigURL` - so whatever you have in book.json for your mappings is used for your old static builds too. No url re-writes or whatever. 
The only complication we will have (which Daniel is resolving) is the redirect from the root dev.px4.io to our default build. 

To be clear, I think what you're doing is cool. If we got to the point where a dynamic build was no longer possible we might even consider moving to it :-)

bys1123, 2018-12-31:
@hamishwillee Thanks for your explain. But my point is lot's of developer can't comfortable visit gitbook.com service that hosting Docs of Dronecode, it's very slow. If it's good, we don't even need to use a mirror site in china.
That's why we(the mirror site) can't use the branch method, I know it's good.

> Incorrect. The selector for all builds is picked up dynamically from what you put in gitbookConfigURL

The examples is using github.io service, so it will redirect to corrected root directory automatically. If use Nginx, It will need to manual config the re-writes.

hamishwillee, 2019-01-01:
@bys1123 that is a good argument for hosting in China, but it is not a good technical argument for using a custom toolchain - there is nothing to stop you using what we publish out of the box hosted on your side of the GFW.

FYI, we are using branches for releases (you can see v1.8.0 and v1.8.2 have now appeared in github). The version support should all be rolled out this week. 

So I see the new generator as cool but unnecessary work "right now".

SHould I close this?

bys1123, 2019-01-05:
I didn't see this project before, https://github.com/PX4/dev.px4.io , this fixed my problem.

bys1123, 2019-01-07:
@hamishwillee Thanks for reminding me. I think this problem could use Nginx's `sub_filter` module easy to fix. I'll let you know if I had a real problem. :-)