PX4/PX4-Devguide #685: add xxd via vim as a dep for ubuntu?

alecGraves, 2018-12-06:
Put ```apt-get install vim``` in ubuntu_sim_nuttx.sh

like with  PR #301 

I was building on docker and got error that xxd was not installed:

```bash
make px4fmu-v5_default
-- PX4 VERSION: v1.8.0-1490-gf637f6a
# BLAH BLAH BAH ...
-- ROMFS: px4fmu_common
CMake Error at ROMFS/CMakeLists.txt:157 (message):
  xxd not found
```

as the msg implies, xxd is used to do something in ROMFS/CMakeLists.txt

```CMake
# create nsh_romfsimg.c
find_program(XXD xxd)
if(NOT XXD)
    message(FATAL_ERROR "xxd not found")
endif()
add_custom_command(OUTPUT nsh_romfsimg.c
        COMMAND ${CMAKE_COMMAND} -E remove -f nsh_romfsimg.c
        COMMAND ${XXD} -i romfs.img nsh_romfsimg.c
        COMMAND sed 's/unsigned/const unsigned/g' nsh_romfsimg.c > nsh_romfsimg.c.tmp && ${CMAKE_COMMAND} -E rename nsh_romfsimg.c.tmp nsh_romfsimg.c
        DEPENDS romfs.img
        )
```

So after a google search, I find out it is distributed with apt in ```vim-common```.

I install vim and now it compiles.

Another solution would be to switch to something more common like ```od```?

Anyway, maybe this issue (/ pr if someone makes one) might help someone someday

alecGraves, 2018-12-07:
Sorry if there is any confusion, but I am not using the official px4 docker image, just the base ubuntu:16.04 image from dockerhub.
* I installed the deps from the ubuntu convenience scripts "ubuntu_sim_nuttx.sh"
* vim was missing

This will probably not be an issue for most users of ununtu, but if your distro for some reason is missing vim, the firmware build will not work.

dagar, 2018-12-10:
I believe that's the easiest way on ubuntu.

PR #689 (2018-12-11): Add vim-common for xxd
Fixes #685
closed 2018-12-11 by hamishwillee