PX4/PX4-Devguide #696: Doc Bug: Building the Code · PX4 Developer Guide

seanrowens, 2018-12-14:
DESCRIBE PROBLEM WITH DOCS HERE

Bug Page: [Building the Code · PX4 Developer Guide](https://dev.px4.io/en/setup/building_px4.html)

On OS X 10.14.1 but also on prior version of OS X, the instructions say to execute "make px4_sitl jmavsim", which does not work;

> SeanOwensComputer:Firmware sean$ make px4_sitl jmavsim
> Makefile:421: *** "px4_sitl cannot be the first argument. Use '/Applications/Xcode.app/Contents/Developer/usr/bin/make help|list_config_targets' to get a list of all possible [configuration] targets.".  Stop.
> SeanOwensComputer:Firmware sean$

Executing "make posix jmavsim" does work, and brings up jmavsim as well as the px4 commander.  

Executing "make list_config_targets" returns a list that does not include anything containing the string "jmavsim".   There are a variety of targets that contain the string posix, but none of them are simply "posix".

mrpollo, 2018-12-14:
@hamishwillee is this due to https://github.com/PX4/Firmware/pull/10875 ?

hamishwillee, 2018-12-15:
@seanrowens I'm closing this. The docs reflect, as they have always, the master build. That buggers things up when we have large user-facing changes. The solution is https://github.com/PX4/Devguide/issues/676.

@mrpollo Yes. 
@dagar Can I please have an ETA on work for version builds. This is coming up almost every day from someone.

closed 2018-12-15 by hamishwillee

hamishwillee, 2018-12-17:
FYI Daniel is away, but we hope to look into this next week (fitting in around Xmas)