PX4/PX4-Devguide #740: Doc Bug: Contributing & Dev Call · PX4 Developer Guide

mrpollo, 2019-02-06:
Dev call format, time and meeting link changed, I suspect this will keep changing we need to find a way to represent this data more accurately if possible.

New link: https://zoom.us/j/946175205
New time: **5:00pm - 6:30pm** (45min each)

Should probably also include the #dev-call channel on Slack

Bug Page: [Contributing & Dev Call · PX4 Developer Guide](https://dev.px4.io/en/contribute/)

hamishwillee, 2019-02-07:
FYI, we have a bug that the root developer guide which "looks like" master, is not updated. I am hoping that Dagar can fix this, but what it means is that the actual current docset is https://dev.px4.io/master/en/contribute/

hamishwillee, 2019-02-07:
@dagar ^^^^ is going to cause problems, because things like devcall info will appear to be old, but are just out of date.

hamishwillee, 2019-02-07:
@mrpollo I'm going to close this because the link is now fix. The start time is also correct. See https://dev.px4.io/master/en/contribute/dev_call.html#schedule

closed 2019-02-07 by hamishwillee