PX4/PX4-Devguide #747: Add PlotJuggler to the log analysis docs

MaEtUgR, 2019-02-21:
Thanks to @bresch for showing me the awsome tool https://www.plotjuggler.io/ which now supports ulog as input format. I just gave it a try and it looks quite awsome as a much more performant and powerful alternative to flightplot.

I think we should add it to the docs to spread the word. When I get to it I can do that myself.

bkueng, 2019-02-21:
Already done! https://github.com/PX4/Devguide/pull/743 :)

MaEtUgR, 2019-02-21:
@bkueng Thanks for mentioning, I don't consider it done yet because:
- I don't know why but you don't find it listed anywhere right now. Shouldn't it be here? https://dev.px4.io/en/log/ulog_file_format.html#known-implementations I searched for `PlotJuggler` in all public docs before creating this issue.
- No screenshot, explanation or alike on https://docs.px4.io/en/log/flight_log_analysis.html
@hamishwillee already mentioned this here: https://github.com/PX4/Devguide/pull/743#issuecomment-464916808

facontidavide, 2019-02-21:
I will prepare a screenshot and some text to add to https://docs.px4.io/en/log/flight_log_analysis.html

facontidavide, 2019-02-21:
[image]

PlotJuggler is a Qt5 desktop application that allows users to easily visualize and analyze data expressed as time series. It supports ULog (.ulg) since version 2.1.3.

Key features:
- Intuitive drag&drop interface.
- Arrange your data in multiple plots, tabs or window.
- Once you arranged your data, save it into a "Layout" file to reload it multiple times.
- Process your data inside PlotJuggler itself, using custom "data transformations".

MaEtUgR, 2019-02-21:
@facontidavide Perfect! Do you want to add the info to https://github.com/PX4/px4_user_guide/blob/master/en/log/flight_log_analysis.md with a pr yourself?

facontidavide, 2019-02-21:
Pull request sent

hamishwillee, 2019-02-22:
@MaEtUgR et al, just FYI, this is in the Ulog page - see https://dev.px4.io/master/en/log/ulog_file_format.html#known-implementations

The problem is with our multi version docs system. Changes to the master branch are supposed to appear both in the root and in the /master tree - but for whatever reason this is not updating the root version. @dagar can you please investigate?

hamishwillee, 2019-02-22:
I'll close this, because there is no more action that is specific to this case.

@facontidavide Thanks for the user guide PR - will review shortly.

closed 2019-02-22 by hamishwillee

MaEtUgR, 2019-02-26:
Thanks, it was here: https://github.com/PX4/px4_user_guide/pull/446 (I like to keep things referenced)