PX4/PX4-Devguide #751: Clarify simulator/ROS and Linux versions

hamishwillee, 2019-03-15:
[Linux · PX4 Developer Guide](http://dev.px4.io/en/setup/dev_env_linux.html) has a number of scripts that were developed on Ubuntu 16.04. I'd like to retest on Ubuntu 18.04 and provide updated/recommended environment.

Three main scripts:
- [ubuntu_sim_nuttx.sh](https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim_nuttx.sh) = nuttx and jmavsim. 
  - Testing shows fine on Ubuntu 18.04
- [ubuntu_sim.sh](https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim.sh) += Gazebo 9. 
  - Testing shows works fine on Ubuntu 18.04.
- [ubuntu_sim_ros_gazebo.sh](https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim_ros_gazebo.sh) = ROS Kinetic with Gazebo 7 default.
  - Fails, log below.

hamishwillee, 2019-03-15:
@TSC21 Can you please confirm:
- Gazebo 9 is still the latest version we support?
- Have we moved on from ROS Kinetic/do we still support it? If so, 
  - what version do we use, 
  - what Gazebo does that use
  - does it work on Ubuntu 16.04, and what Ubuntu is our standard testing environment for that ROS?

I have a log below of the build for Ubuntu 18.04 of our standard script FYI

hamishwillee, 2019-03-15:
@TSC21 On Ubuntu 18.04 below is the build log for our [ubuntu_sim_ros_gazebo.sh](https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim_ros_gazebo.sh)  script. 

Any advice to fix these? Note that the script does run geographiclib setup script as described here: http://dev.px4.io/en/ros/mavros_installation.html#binary-installation-debian--ubuntu

```
Writing /home/ubuntu/catkin_ws/src/.rosinstall

update complete.
usage: rosinstall_generator [-h] [--debug] [--verbose] --rosdistro ROSDISTRO
                            [--from-path [FROM_PATH [FROM_PATH ...]]]
                            [--repos [reponame [reponame ...]]]
                            [--upstream | --upstream-development]
                            [--deps | --deps-up-to [DEPS_UP_TO [DEPS_UP_TO ...]]]
                            [--deps-depth N] [--deps-only]
                            [--wet-only | --dry-only | --catkin-only | --non-catkin-only]
                            [--exclude [EXCLUDE [EXCLUDE ...]]]
                            [--exclude-path [EXCLUDE_PATH [EXCLUDE_PATH ...]]]
                            [--flat] [--tar]
                            [pkgname [pkgname ...]]
rosinstall_generator: error: argument --rosdistro is required
usage: rosinstall_generator [-h] [--debug] [--verbose] --rosdistro ROSDISTRO
                            [--from-path [FROM_PATH [FROM_PATH ...]]]
                            [--repos [reponame [reponame ...]]]
                            [--upstream | --upstream-development]
                            [--deps | --deps-up-to [DEPS_UP_TO [DEPS_UP_TO ...]]]
                            [--deps-depth N] [--deps-only]
                            [--wet-only | --dry-only | --catkin-only | --non-catkin-only]
                            [--exclude [EXCLUDE [EXCLUDE ...]]]
                            [--exclude-path [EXCLUDE_PATH [EXCLUDE_PATH ...]]]
                            [--flat] [--tar]
                            [pkgname [pkgname ...]]
rosinstall_generator: error: argument --rosdistro is required
/usr/lib/python2.7/dist-packages/wstool/config_yaml.py:74: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  yamldata = yaml.load(stream)
Merge caused no change, no new elements found
/usr/lib/python2.7/dist-packages/wstool/config_yaml.py:74: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  yamldata = yaml.load(stream)

Command 'rosdep' not found, but can be installed with:

sudo apt install python-rosdep2

Command 'rosdep' not found, but can be installed with:

sudo apt install python-rosdep2

/usr/lib/python2.7/dist-packages/catkin_tools/config.py:93: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  yaml_dict = yaml.load(f)
-----------------------------------------------------------
Profile:                     default
Extending:                   None
Workspace:                   /home/ubuntu/catkin_ws
-----------------------------------------------------------
Source Space:       [exists] /home/ubuntu/catkin_ws/src
Log Space:         [missing] /home/ubuntu/catkin_ws/logs
Build Space:        [exists] /home/ubuntu/catkin_ws/build
Devel Space:        [exists] /home/ubuntu/catkin_ws/devel
Install Space:      [unused] /home/ubuntu/catkin_ws/install
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
[... 37 lines cut ...]
Failed     << catkin_tools_prebuild:cmake          [ Exited with code 1 ]      
Failed    <<< catkin_tools_prebuild                [ 1.1 seconds ]             
[build] Summary: 0 of 1 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 1.2 seconds total.                                            
bash: /home/ubuntu/catkin_ws/devel/setup.bash: No such file or directory
Downloading dependent script 'install_geographiclib_datasets.sh'
--2019-03-14 18:05:33--  https://raw.githubusercontent.com/mavlink/mavros/master/mavros/scripts/install_geographiclib_datasets.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.192.133, 151.101.128.133, 151.101.64.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.192.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1018 [text/plain]
Saving to: ‘STDOUT’

-                   100%[===================>]    1018  --.-KB/s    in 0s      

2019-03-14 18:05:34 (167 MB/s) - written to stdout [1018/1018]

bash: line 26: hash: geographiclib-get-geoids: not found
bash: line 30: hash: geographiclib-datasets-download: not found
OS not supported! Check GeographicLib page for supported OS and lib versions.
```

TSC21, 2019-03-17:
> @TSC21 Can you please confirm:
> 
> * Gazebo 9 is still the latest version we support?

Gazebo 9 is still the latest version we support yes. 

> * Have we moved on from ROS Kinetic/do we still support it? If so,
>   
>   * what version do we use,
>   * what Gazebo does that use
>   * does it work on Ubuntu 16.04, and what Ubuntu is our standard testing environment for that ROS?

We moved from ROS Kinetic to ROS Melodic. The standard Gazebo version installed with ROS simulation tools is Gazebo 9. And no, ROS Melodic doesn't work under 16.04.

So the standard testing environment will be:
* Ubuntu 16.04 -> ROS Kinetic -> Gazebo 7
* Ubuntu 18.04 -> ROS Melodic -> Gazebo 9

> @TSC21 On Ubuntu 18.04 below is the build log for our [ubuntu_sim_ros_gazebo.sh](https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim_ros_gazebo.sh) script.

That script won't work because you are trying to install ROS Kinetic under Ubuntu 18.04. The supported version is ROS Melodic.

> Any advice to fix these? Note that the script does run geographiclib setup script as described here: http://dev.px4.io/en/ros/mavros_installation.html#binary-installation-debian--ubuntu

Ok let's check this a problem at a time, because I can see there's more than one problem here.
 
> ```
> Writing /home/ubuntu/catkin_ws/src/.rosinstall
>
> update complete.
> usage: rosinstall_generator [-h] [--debug] [--verbose] --rosdistro ROSDISTRO
>                             [--from-path [FROM_PATH [...]

This is failing because the ROS environment is not being sourced correctly somehow. I can see

```
## Setup environment variables
rossource="source /opt/ros/kinetic/setup.bash"
if grep -Fxq "$rossource" ~/.bashrc; then echo ROS setup.bash already in .bashrc;
else echo "$rossource" >> ~/.bashrc; fi
eval $rossource
```

As said, if we are installing under 18.04, ROS Kinetic is not supported. So everything that has a `kinetic` reference needs to be replaced by `melodic`, which is the ROS distro that's supported under Ubuntu 18.04.

> Downloading dependent script 'install_geographiclib_datasets.sh'
> --2019-03-14 18:05:33--  https://raw.githubusercontent.com/mavlink/mavros/master/mavros/scripts/install_geographiclib_datasets.sh [...]

That means that `geographiclib-tools` is not installed. The "simplified" MAVROS install instructions under https://dev.px4.io/en/ros/mavros_installation.html fail to state that one requires to install `geographiclib-tools`, which is something you can find under https://github.com/mavlink/mavros/tree/master/mavros#required-dependencies. Though, this shouldn't be required if we actually have the `rosdep install` working (automatically installs the package dependencies depending on the packages referred in `package.xml` - https://github.com/mavlink/mavros/blob/master/mavros/package.xml#L30), which is not the case since it's failing from the beginning.

hamishwillee, 2019-03-18:
Thanks very much!  As per https://github.com/PX4/Devguide/pull/753 I've updated the script and most of the docs. If you can confirm the build looks good I can close this :-)

PR #753 (2019-03-30): Dev env script and update for Ubuntu 18.04/ROS Melodic
Fixes #751

This is required because PX4 updated to ROS melodic, which only runs on 18.04. 
This will require a little more work, but would be good to get the new script reviewed before we proceed.
closed 2019-03-30 by hamishwillee