PX4/PX4-Devguide #765: Linux toolchain script flaws

MaEtUgR, 2019-04-02:
Since I'm currently setting up a completely new system I wanted to note two problems I faced when using the shell script to install the whole toolchain on Linux https://dev.px4.io/master/en/setup/dev_env_linux.html#pixhawknuttx-and-jmavsim on latest Ubuntu 18.04.2.
- When running the script once something fails and I wasn't able to build the firmware e.g. `make posix` afterwards. I needed to run it again (and then reboot) for it to work. Sadly I didn't capture all the output so I have to repeat the process in a new VM to further analyze what is failing the first time.
- Gazebo doesn't work after using the script. Even simple `gazebo` command fails with
```
maetugr@ubuntuVM:~/Firmware$ gazebo
gazebo: symbol lookup error: /usr/lib/x86_64-linux-gnu/libgazebo_common.so.9: undefined symbol: _ZN8ignition10fuel_tools12ClientConfig12SetUserAgentERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
```
Following https://bitbucket.org/osrf/gazebo/issues/2448/problem-running-gazebo7 I ran
```
sudo apt install libignition-math2
sudo apt upgrade libignition-math2
```
And only after the second command (maybe the first one is not needed) `gazebo` starts and `make posix gazebo` works.

If anyone already knows the proper solution please correct the script if not I'll probably need to try on a VM with snapshots to find a good solution that works in general and hopefully doesn't break again (I know it worked before).

hamishwillee, 2019-04-03:
@MaEtUgR That script doesn't install gazebo, only jmavsim. Hence the name. 

Note, I did run this recently on Ubuntu 18.04 and it appeared to install and run fine.

That said, yes, the scripts are not perfect. I'm hoping to fix this with Daniels script: https://github.com/PX4/Firmware/pull/11768 - basically a script that lives in CI and gets tested regularly.

hamishwillee, 2019-04-03:
Try this one https://dev.px4.io/master/en/setup/dev_env_linux.html#jmavsimgazebo-simulation

MaEtUgR, 2019-04-03:
@hamishwillee oh thanks, jmavsim worked fine after running it for the second time. And I was confused because gazebo gets installed maybe because of ROS or some other dependency but it's not the official gazebo script and I didn't see that thanks. Daniel wrote me about the new firmware versioned scripts shortly after. I think that's the way to go and we should bring them up in a maintainable way instead of wasting too much time on the old ones which still mostly work. I'll close this and try to get in the loop with things I can help on the new scripts and e.g. make sure https://github.com/PX4/Devguide/pull/719/files#diff-57e9a9559326143dea03f6c6031f24d2R25 works.

closed 2019-04-03 by MaEtUgR

julianoes, 2019-04-03:
I just ran this script on 18.04.2 and it ran through fine:
https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim_common_deps.sh

Then this one fails at the very end:
https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim.sh

```
./ubuntu_sim.sh: line 37: cd: /home/julianoes/src/Firmware: No such file or directory
```

Sidenote: I really don't like the fact that a script installing dependencies also clones the repository. It makes it confusing.

MaEtUgR, 2019-04-03:
> I really don't like the fact that a script installing dependencies also clones the repository. It makes it confusing.

Totally agree, such steps should if at all be optional in a different beginners script that can be ran after the pure installation.

julianoes, 2019-04-03:
Also, these scripts should fail on error using `-s`.

MaEtUgR, 2019-04-03:
> `./ubuntu_sim.sh: line 37: cd: /home/julianoes/src/Firmware: No such file or directory`

That's probably it, mine also stopped in the `~/src/` folder when it failed if I remember correctly.

hamishwillee, 2019-04-04:
Thanks @MaEtUgR - Yes, the right thing for us to do is make sure that the in-tree CI-tested scripts do what we need. The current direction is one script that will allow conditional installation to replace this set of scripts.

> Sidenote: I really don't like the fact that a script installing dependencies also clones the repository. It makes it confusing.

@julianoes I've heard that before, and the new scripts won't do this. You'll either download firmware first and run them in-tree, or fetch them from github and separately get the tree. 

"My" scripts do it this way because I'm often running tests on fully clean builds. It saves me time.