PX4/PX4-Devguide #767: Doc Bugv1.8.2: Jenkins Continuous Integration · PX4 v1.8.2 Developer Guide

oneWayOut, 2019-04-02:
Bug Page: [Jenkins Continuous Integration · PX4 v1.8.2 Developer Guide](https://dev.px4.io/en/test_and_ci/jenkins_ci.html)

The links of [run_container.bash](https://github.com/PX4/Firmware/blob/master/integrationtests/run_container.bash) and  [run_tests.bash](https://github.com/PX4/Firmware/blob/master/integrationtests/run_tests.bash) don't exist anymore.

hamishwillee, 2019-04-03:
Thanks @oneWayOut 

@dagar The latest docs on test execution are no longer correct: https://dev.px4.io/master/en/test_and_ci/jenkins_ci.html#test-execution

Can you clarify how this works?

Note, I have updated the old versions so they point to master branch. So in future we just need to keep master up to date.

hamishwillee, 2021-01-28:
Issue moved to user guide: https://github.com/PX4/PX4-user_guide/issues/1019

closed 2021-01-28 by hamishwillee