PX4/PX4-Devguide #768: Removal of modemmanager not needed anymore

julianoes, 2019-04-03:
In the toolchain install scripts we still uninstall the modemmanager packet (without even asking users if they agree with that) although since modemmanager 1.10 Pixhawk-like devices with the USB ID 0x26ax are blacklisted ([see modemmanager patch](https://github.com/freedesktop/ModemManager/commit/9447527c357a91bbd3660274c41fd1b802e4cf84)).
Given Ubuntu 16.04 comes with modemmanager 1.40, it's about time we remove that workaround.

hamishwillee, 2019-04-04:
@julianoes The Devguide _scripts_ don't AFAIK remove the modem manager - but the _docs_ do say that you have to do this. So we could just remove the section from the docs. It appears both on the linux parent page and on the sub page.

Do we still need to add to the dialout group?

@DonLakeFlyer Note above, the check for modemmanager may not be needed on QGC any more.

julianoes, 2019-04-04:
> Do we still need to add to the dialout group?

Yes that's still required to have permission to access the USB devices as far as I know.

julianoes, 2019-04-04:
> @julianoes The Devguide scripts don't AFAIK remove the modem manager 

I think they do:

https://github.com/PX4/Devguide/blob/a6240dc1754605c9feb1892d87a33fff7a500ad7/build_scripts/ubuntu_sim_common_deps.sh#L16-L17

I'll make a PR.

DonLakeFlyer, 2019-04-19:
But that only handles the old 3DR VID. Things like Cube and Holybro new stuff no longer use that VID.

julianoes, 2019-04-19:
@DonLakeFlyer ok good to know. These vendors should make sure to get on the ignore list then, and until then we'll need to leave the warning.

And it's probably a good idea to test if it's actually still a problem or if modemmanager still blocks any /dev/ttyACM*.

hamishwillee, 2019-05-01:
@julianoes FMI, so the summary is that we ignore this for now?

julianoes, 2019-05-01:
Ok, fair.

closed 2019-05-01 by julianoes