PX4/PX4-Devguide #773: URL for branches not explicit enough

julianoes, 2019-04-26:
I would like the URL to contain the version it refers to. I don't like the fact that a URL pointing to docs about 1.8.2 does not contain 1.8.2. This means the links won't be stable over time and can break, e.g. when 1.9.0 is the new stable.

Current state:
https://dev.px4.io/en/advanced/parameter_reference.html  -> 1.8.2
https://dev.px4.io/master/en/advanced/parameter_reference.html -> master

I'd suggest the pattern as follows:
https://dev.px4.io/1.8.2/en/advanced/parameter_reference.html -> 1.8.2
https://dev.px4.io/master/en/advanced/parameter_reference.html -> master

The default redirect when visiting the page can still be 1.8.2.

Is this possible @hamishwillee?

hamishwillee, 2019-04-29:
@julianoes Every branch DOES get a proper URL exactly as you have suggested. So there is
- https://dev.px4.io/master/en/advanced/parameters_and_configurations.html  
- https://dev.px4.io/v1.8.2/en/advanced/parameters_and_configurations.html  
- https://dev.px4.io/v1.8.0/en/advanced/parameters_and_configurations.html

In the root (dev.px4.io/) we have "stable" which is currently v1.8.2 but over time will be 1.9.x or whatever. This is a manual copy of whatever is in the current stable branch - so in this case I copied the 1.8.2 docs.

You are correct that links to stable will break over time. That is also true to links to master, as master changes. 

What would be better than what we do now is to have a redirect from https://dev.px4.io/ to stable rather than duplication. So if you arrive anywhere in the root that isn't an actual branch you get sent to the proper URL for the branch. The problem here is that we're hosting in github and that does not allow us to define redirects via regex as we could if we self hosted. 
Another option would be to create manual redirects in the root so links to "stable" end up in the right place, but @dagar thought this not worth the effort - and I somewhat agree (would also be impossible to do robustly).

Upshot, to fix this we would have to properly self host. You'd have to argue that with @dagar.

julianoes, 2019-04-29:
> So if you arrive anywhere in the root that isn't an actual branch you get sent to the proper URL for the branch.

Exactly but I understand the challenge. 

Knowing that I can share links with `v1.8.2` I'll try to do that from now on, in the same way that I try to get "permanent links" when linking to source on GitHub.

hamishwillee, 2019-04-30:
Great. Wish this could be more robust, but realistically, it won't change anytime soon. So closing.

closed 2019-04-30 by hamishwillee