PX4/PX4-Devguide #782: jmavsim: WARNING: An illegal reflective access operation has occurred:

CarlosFernandoTovarCeron, 2019-05-06:
I am new and trying run python scrips for dronecode sdk.

I follow this guide: https://dev.px4.io/en/setup/dev_env_linux.html

I install jMAVSim/Gazebo with ubuntu_sim.sh.

I am using 1.9.0 branch (HEAD detached at v1.9.0-rc0)

When I try make "make posix jmavsim", GUI fails and show me WARNINGS.

I am using ubuntu 18.10 and i try install jdk 11 with "sudo apt-get install default-jdk"

Thanks for your help!
`
almendroavellano@hp-pavilion-gaming-laptop:~/src/Firmware$ make posix jmavsim
[0/1] cd /home/almendroavellano/src/Firmware/buil...endroavellano/src/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/almendroavellano/src/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
src_path: /home/almendroavellano/src/Firmware
build_path: /home/almendroavellano/src/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/home/almendroavellano/src/Firmware/build/px4_sitl_default/bin/px4" "/home/almendroavellano/src/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/almendroavellano/src/Firmware"/test_data
INFO  [px4] Creating symlink /home/almendroavellano/src/Firmware/ROMFS/px4fmu_common -> /home/almendroavellano/src/Firmware/build/px4_sitl_default/tmp/rootfs/etc
0 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
  SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1376264
  CAL_ACC1_ID: curr: 0 -> new: 1310728
  CAL_ACC_PRIME: curr: 0 -> new: 1376264
  CAL_GYRO0_ID: curr: 0 -> new: 2293768
  CAL_GYRO_PRIME: curr: 0 -> new: 2293768
  CAL_MAG0_ID: curr: 0 -> new: 196616
  CAL_MAG_PRIME: curr: 0 -> new: 196616
  COM_DISARM_LAND: curr: -1.0000 -> new: 0.1000
  COM_OBL_ACT: curr: 0 -> new: 2
  COM_OF_LOSS_T: curr: 0.0000 -> new: 5.0000
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_I: curr: 0.0200 -> new: 0.1500
  MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
  MPC_XY_P: curr: 0.9500 -> new: 0.8000
  MPC_XY_VEL_P: curr: 0.0900 -> new: 0.2000
  MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0160
  MPC_JERK_MIN: curr: 8.0000 -> new: 10.0000
  MPC_ACC_HOR_MAX: curr: 10.0000 -> new: 3.0000
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  NAV_DLL_ACT: curr: 0 -> new: 2
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 3 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
* RTL_DESCEND_ALT: curr: 5.0000 -> new: 10.0000
* RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
Buildfile: /home/almendroavellano/src/Firmware/Tools/jMAVSim/build.xml
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Inconsistency detected by ld.so: dl-lookup.c: 112: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!
`

CarlosFernandoTovarCeron, 2019-05-06:
This is my current java
openjdk 11.0.2 2019-01-15
OpenJDK Runtime Environment (build 11.0.2+9-Ubuntu-3ubuntu118.10.3)
OpenJDK 64-Bit Server VM (build 11.0.2+9-Ubuntu-3ubuntu118.10.3, mixed mode, sharing)

hamishwillee, 2019-05-06:
@CarlosFernandoTovarCeron Might make no difference but the docs for 1.9 are in the "master" tree: https://dev.px4.io/master/en/

Might well be a java version issue. The scripts install openjdk-8-jdk openjdk-8-jre, while the docker used for testing installs default-jre-headless. You can see what docker does currently in these scripts: https://github.com/PX4/Firmware/tree/master/Tools/setup

We hope to move to the docker scripts for end users sometime soon. They need more testing.

CarlosFernandoTovarCeron, 2019-05-06:
Thank you, should I install requirements.txt in an env?

CarlosFernandoTovarCeron, 2019-05-06:
Hi, I remove my java and I install with:
`sudo apt install openjdk-8-jdk openjdk-8-jre`

Now, java --version show: 
`openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.18.10.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
`
And I have a different error:
`almendroavellano@hp-pavilion-gaming-laptop:~/src/Firmware$ java -version
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.18.10.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
almendroavellano@hp-pavilion-gaming-laptop:~/src/Firmware$ make posix jmavsim
[0/1] cd /home/almendroavellano/src/Fi...no/src/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/almendroavellano/src/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
src_path: /home/almendroavellano/src/Firmware
build_path: /home/almendroavellano/src/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/home/almendroavellano/src/Firmware/build/px4_sitl_default/bin/px4" "/home/almendroavellano/src/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/almendroavellano/src/Firmware"/test_data
INFO  [px4] Creating symlink /home/almendroavellano/src/Firmware/ROMFS/px4fmu_common -> /home/almendroavellano/src/Firmware/build/px4_sitl_default/tmp/rootfs/etc
0 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
  SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1376264
  CAL_ACC1_ID: curr: 0 -> new: 1310728
  CAL_ACC_PRIME: curr: 0 -> new: 1376264
  CAL_GYRO0_ID: curr: 0 -> new: 2293768
  CAL_GYRO_PRIME: curr: 0 -> new: 2293768
  CAL_MAG0_ID: curr: 0 -> new: 196616
  CAL_MAG_PRIME: curr: 0 -> new: 196616
  COM_DISARM_LAND: curr: -1.0000 -> new: 0.1000
  COM_OBL_ACT: curr: 0 -> new: 2
  COM_OF_LOSS_T: curr: 0.0000 -> new: 5.0000
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_I: curr: 0.0200 -> new: 0.1500
  MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
  MPC_XY_P: curr: 0.9500 -> new: 0.8000
  MPC_XY_VEL_P: curr: 0.0900 -> new: 0.2000
  MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0160
  MPC_JERK_MIN: curr: 8.0000 -> new: 10.0000
  MPC_ACC_HOR_MAX: curr: 10.0000 -> new: 3.0000
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  NAV_DLL_ACT: curr: 0 -> new: 2
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 3 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
* RTL_DESCEND_ALT: curr: 5.0000 -> new: 10.0000
* RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
Buildfile: /home/almendroavellano/src/Firmware/Tools/jMAVSim/build.xml
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: java.awt.AWTError: Assistive Technology not found: org.GNOME.Accessibility.AtkWrapper
	at java.awt.Toolkit.loadAssistiveTechnologies(Toolkit.java:807)
	at java.awt.Toolkit.getDefaultToolkit(Toolkit.java:886)
	at java.awt.Window.getToolkit(Window.java:1358)
	at java.awt.Window.init(Window.java:506)
	at java.awt.Window.<init>(Window.java:537)
	at java.awt.Frame.<init>(Frame.java:420)
	at java.awt.Frame.<init>(Frame.java:385)
	at javax.swing.JFrame.<init>(JFrame.java:189)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:107)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:171)
	at me.drton.jmavsim.Simulator.main(Simulator.java:824)
	... 5 more

`

hamishwillee, 2019-05-06:
That could be a driver issue, for which I have no solution. @julianoes any ideas?

If you don't strictly speaking need jmavsim you might want to try Gazebo headless:
https://dev.px4.io/master/en/simulation/gazebo.html#headless-mode

julianoes, 2019-05-06:
After swapping the java version, have you made sure to rebuild jMAVSim. To clean up you have to do:
```
rm -rf Tools/jMAVSim/out
```

CarlosFernandoTovarCeron, 2019-05-06:
thanks @hamishwillee  and @julianoes, I was looking Gazebo simulator and feels better, I will look in detail

To solve, I do this:
follow this tutorial for manual oracle java 8 installation: [https://www.oracle.com/technetwork/java/javase/downloads/jdk8-downloads-2133151.html](url)
run `sudo apt-get install ant`
run `rm -rf Tools/jMAVSim/out`
run `make distclean`
run `make posix jmavsim`

julianoes, 2019-05-06:
Ok. I wonder if this error is responsible for your problem:
```
 Caused by: java.awt.AWTError: Assistive Technology not found: org.GNOME.Accessibility.AtkWrapper
```
https://askubuntu.com/questions/695560/assistive-technology-not-found-awterror

pranav083, 2019-10-07:
> thanks @hamishwillee and @julianoes, I was looking Gazebo simulator and feels better, I will look in detail
> 
> To solve, I do this:
> follow this tutorial for manual oracle java 8 installation: https://www.vultr.com/docs/how-to-manually-install-java-8-on-ubuntu-16-04
> run `sudo apt-get install ant`
> run `rm -rf Tools/jMAVSim/out`
> run `make distclean`
> run `make posix jmavsim`

On the website https://www.vultr.com/docs/how-to-manually-install-java-8-on-ubuntu-16-04 . Don't forget to use the upadted file version of downloaded jdk file . As  while installing it will return error.

hamishwillee, 2019-10-07:
@julianoes This problem has existed for me when running in a VM for a couple of years - ie since at least PX4 v 1.7 and on both VMWare and Virtual Box, for Ubuntu 16.04 and Ubuntu 18.04. It occurs for both our old and new installer scripts. It does not occur on my real linux computer (Ubuntu 16.04 - setup with the same scripts). 

It does not occur if you run Jmavsim headless - which is now my preferred setup. 

Anyway, do we know that this approach outlined works - ie using manual installation of oracle jdk8 rather than openjdk as per our scripts. If it does, are there reasons not to switch?

pranav083, 2019-10-07:
https://github.com/PX4/Firmware/issues/12385 . This is the same error i am facing when i am building the file for ros - gazebo simulation . And it has something to do with the "protoc " libraries .

pranav083, 2019-10-07:
their is no folllow up their on this thread . And i dont know how to solve this issue.

> [PX4/Firmware#12385](https://github.com/PX4/Firmware/issues/12385) . This is the same error i am facing when i am building the file for ros - gazebo simulation . And it has something to do with the "protoc " libraries .

hamishwillee, 2019-10-07:
@pranav083 Please don't post unrelated issues on other threads.  I have deleted the comments.

julianoes, 2019-10-08:
> If it does, are there reasons not to switch?

I don't know.

hamishwillee, 2020-11-13:
Pretty sure this is sorted now.

closed 2020-11-13 by hamishwillee