PX4/PX4-Devguide #783: remap from orb_slam2_stereo to mavros/vision_pose/pose

guoxianzhen, 2019-05-07:
I use orb_slam2_ros package which publishes position information of copter on topic "orb_slam2_stereo/pose". The type of these information is geometry/PostStamped. I want to remap this topic to "mavros/vision_pose/pose". So I added a remap line () in the orb_slam2_stereo.launch. But the remap didn't work. How can i make it work. Have I added the remap line in a wrong place?
This is the orb_slam2_stereo.launch
<launch>
  <node name="orb_slam2_stereo" pkg="orb_slam2_ros"
      type="orb_slam2_ros_stereo" args="
          $(find orb_slam2_ros)/orb_slam2/Vocabulary/ORBvoc.txt
          $(find orb_slam2_ros)/orb_slam2/config/stereo.yaml"
      output="screen">
       <remap from="image_left/image_color_rect" to="/stereo/left/image_color" />
       <remap from="image_right/image_color_rect" to="/stereo/right/image_color" />
       <remap from="/mavros/vision_pose/pose" to="/orb_slam2_stereo/pose"/>

       <param name="publish_pointcloud" type="bool" value="true" />
       <param name="publish_pose" type="bool" value="true" />
       <param name="pointcloud_frame_id" type="string" value="fcu" />
       <param name="camera_frame_id" type="string" value="camera_link" />
       
  </node>
</launch>
when I try to echo the information from "mavros/vision_pose/pose", the output is rostopic echo mavros/vision_pose/pose WARNING: no messages received and simulated time is active. Is /clock being published?

guoxianzhen, 2019-05-07:
I added a <remap from="/mavros/vision_pose/pose" to="/orb_slam2_stereo/pose"/> in the orb_slam2_stereo.launch

hamishwillee, 2019-05-08:
@guoxianzhen This is not the right place to ask support questions. The right thing to do is raise this in the discussion boards: https://discuss.px4.io/c/software-development

If at some point you then discover a problem with the PX4 Devguide documentation this would be the right place to post it.

closed 2019-05-08 by hamishwillee