PX4/PX4-Devguide #790: Define the format of ULog file using XML,  which used in MAVLink.

Marxlp, 2019-05-15:
After Reading the the [MAVLink devguide](https://mavlink.io/en/), I realized that ULog file structure can be described, just like `MAVLink`, in `XML`. Below is my skethy version of the ULog definition in `XML`.
```xml
<?xml version="1.0" encoding="ISO-8859-1"?>
<ULog global_parse="ordered">
    <header>
        <filemagic>
            <size>7</size>
            <validvalue>554c6f6701</validvalue>
        </filemagic>
        <version>
            <size>1</size>
            <validvalue>01</validvalue>
        </version>
        <timestamp>
            <dtype>uint64_t</dtype>
            <size>8</size>
            <value>unknown</value>
            <unit>ms</unit>
        </timestamp>
    </header>
    <definitions parse='causal'>
        <message_header_s>
            <element>
                <name>msg_size</name>
                <dtype>uint16_t</dtype>
            </element>
            <element>
                <name>msg_type</name>
                <dtype>uint8_t</dtype>
            </element>
        </message_header_s>
        <B alias='ulog_message_flag_bits_s'>
            <include>message_header_s</include>
            <element>
                <name>compat_flags</name>
                <dtype>uint8_t</dtype>
                <size>8</size>
            </element>
            <element>
                <name>incompat_flags</name>
                <dtype>uint8_t</dtype>
                <size>8</size>
            </element>
            <element>
                <name>appended_offsets</name>
                <dtype>uint64_t</dtype>
                <size>3</size>
            </element>
        </B>
        <F alias='message_format_s'>
            <include>message_header_s</include>
            <element>
                <name>format</name>
                <size>
                    <left>1</left>
                </size>
            </element>
        </F>
        <I alias='message_info_s'>
            <include>message_header_s</include>
            <element>
                <name>key_len</name>
                <dtype>uint8_t</dtype>
            </element>
            <element>
                <name>key</name>
                <dtype>char</dtype>
                <size>
                    <valueof>key_len</valueof>
                </size>
            </element>
            <element>
                <name>value</name>
                <dtype>char</dtype>
                <size>
                    <left>1</left>
                </size>
            </element>
        </I>
        <M alias='ulog_message_info_multiple_header_s'>
            <include>message_header_s</include>
            <element>
                <name>is_continued</name>
                <dtype>uint8_t</dtype>
            </element>
            <element>
                <name>key_len</name>
                <dtype>uint8_t</dtype>
            </element>
            <element>
                <name>key</name>
                <dtype>char</dtype>
                <size>
                    <valueof>key_len</valueof>
                </size>
            </element>
            <element>
                <name>value</name>
                <dtype>char</dtype>
                <size>
                    <left>1</left>
                </size>
            </element>
        </M>
    </definitions>
    <data_section parse='causal'>
        <A alias='message_add_logged_s'>
            <include>definitions.msg_header_s</include>
            <element>
                <name>multi_id</name>
                <dtype>uint8_t</dtype>
            </element>
            <element>
                <name>msg_id</name>
                <dtype>uint16_t</dtype>
            </element>
            <element>
                <name>message_name</name>
                <dtype>char</dtype>
                <size>
                    <left>1</left>
                </size>
            </element>
        </A>
        <R alias='message_remove_logged_s'>
            <include>definitions.msg_header_s</include>
            <element>
                <name>msg_id</name>
                <dtype>uint16_t</dtype>
            </element>
        </R>
        <D alias='message_data_s'>
            <include>definitioons.msg_header_s</include>
            <element>
                <name>msg_id</name>
                <dtype>uint16_t</dtype>
            </element>
            <element>
                <name>data</name>
                <dtype>uint8_t</dtype>
            </element>
        </D>
        <L alias='message_logging_s'>
            <include>definitions.msg_header_s</include>
            <element>
                <name>log_level</name>
                <dtype>uint8_t</dtype>
            </element>
            <element>
                <name>timestamp</name>
                <dtype>uint64_t</dtype>
            </element>
            <element>
                <name>message</name>
                <dtype>char</dtype>
                <size>
                    <left>1</left>
                </size>
            </element>
        </L>
        <S alias='message_sync_s'>
            <include>definitions.msg_header_s</include>
            <element>
                <name>sync_magic</name>
                <dtype>uint8_t</dtype>
                <size>8</size>
            </element>
        </S>
        <O alias='message_dropout_s'>
            <include>definitions.msg_header_s</include>
            <element>
                <name>duration</name>
                <dtype>uint16_t</dtype>
            </element>
        </O>
        <include>definitions.I</include>
        <include>definitions.M</include>
        <include>definitions.P</include>
    </data_section>
</ULog>
```

Marxlp, 2019-05-15:
To verify and test the idea, I wrote a parser to unpack data according to the XML definitions. The core part is shown below. For details, see [here](https://github.com/Marxlp/px4insight/blob/master/ULog/ULog_Analysis.ipynb).
```python
class ULogUnpacker:
    def __init__(self, definition_xml, ULog_file):
        self.definition_xml = definition_xml
        self.xml_tree = ET.fromstring(definition_xml)
#         self.xml_tree = ET.parse(definition_xml)
        self._sections = ['header', 'definitions', 'datasection']
        self.check_xml_sections()
        self.ULog_file = ULog_file
        self.data = open(self.ULog_file, 'rb').read()
        # parse read place index
        self.index = 0
        self.max_index = len(self.data)
        self.ulog_data = {}
    
    def check_xml_sections(self):
        # check whether the xml include three defined sections
        section_tags = [section.tag for section in self.xml_tree]
        if section_tags != self._sections:
            raise IOError('The structure of the xml definition file is invalid for ULog file. Check the xml structure. It must include ' +
                          ','.join(self._sections[:-1] + ' and' + self._section[-1]))
    
    def _unpack_header(self):
        header_data = self._unpack_sub_section(self.xml_tree[0])
        self.ulog_data['header'] = header_data
    
    def _unpack_definitions(self):
        definitions = []
        self.ulog_data['definitions']  = definitions
        while True:
            msg_type = self._probe_message_type()
            # check whether definitions section ends and data section starts
            if msg_type in ['A','L']:
                break
            msg_definition = self.xml_tree.find('.//' + msg_type)
            definitions.append(self._unpack_sub_section(msg_definition))
            
    def _unpack_data(self):
        data = []
        self.ulog_data['data'] = data
        while True:
            msg_type = self._probe_message_type()
            msg_definition = self.xml_tree.find('.//' + msg_type)
            data.append(self._unpack_sub_section(msg_definition))
            if self.index >= self.max_index:
                break
        
            
    def unpack_ulog(self):
        self._unpack_header()
        self._unpack_definitions()
        self._unpack_data()
        # TODO: process the unpacked data to eleminate some auxiliary info and tranfer
        #      data to readable format.
        
    def _probe_message_type(self):
        message_header_s = self.xml_tree.find('.//message_header_s')
        _index = self.index
        message_header_s_data = self._unpack_sub_section(message_header_s)
        self.index = _index
        return chr(message_header_s_data['msg_type'][0])
        
    @staticmethod
    def _parse_element(element):
        straight_content = {'name', 'dtype', 'validvalue', 'unit'}
        element_structure = {}
        for item in straight_content:
            a = element.find(item)
            if a is not None:
[... 72 lines cut ...]
#                     print(element_structure)
                    type_size = dtype_size[element_structure['dtype']]
                    element_structure['size'] = int((sub_section_size - content_size)/type_size)
                else:
                    # process the reference in size tag
                    if isinstance(element_structure['size'], str):
                        # here we simplify the process of relative reference. Since the value reference is
                        # limited in the subsection struct. it can alway be found in the sub_section_content
#                         pdb.set_trace()
                        element_structure['size'] = self._unpack_reference_value(element_structure['size'], sub_section_content)[0]
#                 pdb.set_trace()
                data_size = element_structure['size']
                type_size = dtype_size[element_structure['dtype']]
                data_section_size = data_size * type_size
                content_size += data_section_size
                element_data = self._unpack_element(self.data[self.index:self.index + data_section_size], element_structure)
                print('element_data:',element_data)
                # check whether the data value is equal to valid value
                if 'validvalue' in element_structure:
                    if element_structure['validvalue'] != list(element_data.values())[0]:
                        raise ValueError('The value in the data is invalid, the valid data is', element_structure['validvalue'])
                self.index += data_section_size
                sub_section_content.update(element_data)
        return sub_section_content
```

hamishwillee, 2019-05-15:
@Marxlp YOu've posted this into the devguide. While I think it is cool, what is the intent/proposal you are making. I what do you think we should do with this?

@bkueng FYI

Marxlp, 2019-05-16:
I didn't know where it should be posted, Firmware or here. It's like a standard or protocol. For my understanding, Firmware is the implementation of the logics in standards, protocols or algorithms. At the same time, in the devguide doc, the data was described by C struct, which I thought is some language-speciafic and an abstractation can be taked base on that. So these are the reasons why I post it here.

Marxlp, 2019-05-16:
As far as I am concerned, the ULog section in the devguide could describe the ULog file struct in XML-like syntax and put  explainations of some key points aside by that.

hamishwillee, 2019-05-16:
Thanks @Marxlp - The data is described by C struct because that is how it is _generated_. That makes it easy to generate, keep in sync with the code, and for people to see how things work. 

I completely agree that the XML representation more as a descriptive and generic format for people developing a parser. My concern is keeping it maintained and in sync with the code. It is already hard to keep this doc in sync with the code.

@bkueng As the person most likely to have to maintain this, what are your thoughts?

Marxlp, 2019-05-16:
Yes, I agree. Sometimes, it really takes time to maintain the doc.  But I didn't mean to have to maintain it. 😅 .

bkueng, 2019-05-16:
I think it's a nice example, and a formal XML representation is useful (however you also need to make it clear how to interpret it).
Given the following:
- People have successfully implemented ULog parsers based on current docs in different languages
- XML is hard to read and you still need additional information to implement a parser
I don't see the benefit of the XML over the current status. I would however see it as useful if we could use it to validate the correctness of a parser/logger.

Marxlp, 2019-05-16:
Well, I agree the XML looks verbose and sometimes is hard to read for people who is not familiar with it. There are still some data format like `json` and `Protobuf`, which are more readable. The main idea behind my proposal is to automate the generation of the parser according to formal definition,  just like what have been done in MAVLink.

hamishwillee, 2019-05-17:
@Marxlp @bkueng Finally, I understand what you mean:-). You're imagining a scenario where the XML defines the format of a log parser, and we have a generator toolchain that can generate the library for parsers in different programming languages: Python, C, or whatever. That way if someone changes the XML all parsers automatically get updated.

How useful it is depends a bit on how often the format gets updated, and whether it the generator infrastructure makes it easier to create a parser than writing one from scratch.

Not sure if it is something that PX4 should specifically enable. Either way though, it could be an independent project - people want to support this idea then it will grow, if not die.
@bkueng Further thoughts

What is clear is that this is not a docs question - ie not something I can usefully contribute to.

bkueng, 2019-05-17:
Yes we can keep it as separate repo, and we can gladly link from the docs to it.
While I like the idea, I think it's hard to do in practise for this use-case, as @hamishwillee mentions, format or logic updates require deeper changes.
This is different from MAVLink, which has a set of enum's and messages that get updated, the rest stays. And a MAVLink version update from 1.0 to 2.0 required updating all language generators as well.

Marxlp, 2019-05-17:
Yes, I agree. Actually, it's some kind of work on the compiler. I will have a try. If I do work out something, I will give you a message.

closed 2019-05-17 by Marxlp