PX4/PX4-Devguide #806: Sending custom message from MAVROS to PX4 documentation example

RicardoM17, 2019-05-29:
I apologise if this is not the correct place but I found a great example that could be added to the official documentation. It can be seen in [this link](https://github.com/JoonmoAhn/Sending-Custom-Message-from-MAVROS-to-PX4/issues/1#issue-252489957). I went through it and everything is running smoothly.

It isn't exactly copy paste as some code needs to change as it is almost 2 years old but ~90% of it is plug and play and the rest is only a few minor changes.

mhkabir, 2019-05-29:
Let's make a page on PX4 Devguide documenting this.

cc @hamishwillee

hamishwillee, 2019-05-29:
@RicardoM17 This is exactly the right place to post, and thanks very much for doing so. We can't just copy it because we need to know the licensing. I have asked @JoonmoAhn for permission (or ideally to add it in a PR).

Would you be able to do the updates to ROS Melodic, PX4 1.9 and Ubuntu 18.04 if it gets submitted?

hamishwillee, 2019-06-03:
OK, the current version is in https://github.com/PX4/Devguide/pull/810 - @RicardoM17 can you ETA being able to update it?

RicardoM17, 2019-06-03:
Also answered on #810 

Yea I can update it. My set up is like you mentioned Melodic, PX4 1.9 Ubuntu 18.04 so once I have something I'll let you know for a review ;)

PR #810 (2019-06-06): Add sending custom messages to PX4 from MAVROS
Kindly contributed by @JoonmoAhn in https://github.com/JoonmoAhn/Sending-Custom-Message-from-MAVROS-to-PX4/issues/1

This has not yet been updated to recent versions.

Fixes #806 

@RicardoM17 can you update to the current dev environment? You can edit this branch directly?
closed 2019-06-06 by hamishwillee