PX4/PX4-Devguide #808: MPC Calibration

albert21martins, 2019-05-30:
Hi,

I think it would be good to get a guide on how to calibrate the MPC. We have a 5Kg Octocopter Coax with Pixhawk 4, which flies really good in Acro, Stabilize, Alttitude and Position when there is no Wind. The Problem is when there is Wind (10Kmh) and in Position mode, the copter starts to get Unstable, corrections are strong and it make it a little insecure to fly. I imagine this is related to MPC. Is there a way to calibrated right?

hamishwillee, 2019-05-30:
You mean like this guide? https://docs.px4.io/en/config_mc/advanced_mc_position_tuning.html#position_mode

bkueng, 2019-06-03:
@albert21martins you are correct, PID tuning for the velocity/position controllers are still missing. @bresch didn't you want to write something?

bresch, 2019-06-03:
@bkueng Yes, I'm trying to find an easy and systematic way to tune the velocity PID loops. I'll write a document for the user/dev guide as soon as I have a good protocol.

hamishwillee, 2019-06-03:
@bkueng @bresch Thanks for the correction. Just FMI/to confirm 
- [Advanced Multicopter Position Control Tuning](https://docs.px4.io/en/config_mc/advanced_mc_position_tuning.html) covers the value of the setpoint rather than tracking of the setpoint (which is what the PID does)
- [Multicopter PID Tuning Guide](https://docs.px4.io/en/config_mc/pid_tuning_guide_multicopter.html) - covers rate and attitude controller tuning, but not velocity/position controllers.

Correct?
-

bkueng, 2019-06-04:
Correct. 'Advanced Multicopter Position Control Tuning' is changing the user experience: how fast you want it to react, the max velocity etc.

albert21martins, 2019-06-05:
yes what I was looking was for a tuning guide for MPC in velocity and Position mode, the outer P and PID. I belive the parameter in question are the following:
- MPC_XY_P
- MPC_XY_VEL_P
- MPC_XY_VEL_D
- MPC_XY_VEL_I
- MPC_Z_P
- MPC_Z_VEL_P
- MPC_Z_VEL_D
- MPC_Z_VEL_I

@bkueng do you have some sketch we could test.

bkueng, 2019-06-06:
Yes these are the right parameters.
Reassigning this to @bresch as he is looking into this.

albert21martins, 2021-01-13:
Hello,
is there any news regarding this topic? I did not find anything yet about it and I think it is an important topic in order to get the best flight performance out of the px4.

hamishwillee, 2021-01-14:
@bresch You stated in 2019 above "Yes, I'm trying to find an easy and systematic way to tune the velocity PID loops. I'll write a document for the user/dev guide as soon as I have a good protocol."

Have you been able to do that? Is there a timeline - and for docs?

kuiskas, 2021-01-25:
I just saw, Auterion has implemented an autotune for the pid's I suppose this is not going to be merged in here or am I wrong?

bresch, 2021-01-25:
> I just saw, Auterion has implemented an autotune for the pid's I suppose this is not going to be merged in here or am I wrong?

@kuiskas Auterion will actually upstream that feature later this year so the whole community can benefit from it :)

> @bresch You stated in 2019 above "Yes, I'm trying to find an easy and systematic way to tune the velocity PID loops. I'll write a document for the user/dev guide as soon as I have a good protocol."
Have you been able to do that? Is there a timeline - and for docs?

Unfortunately not. Manual tuning of the velocity loops is quite difficult without seeing the plots in real time. However, the tuning page of QGC is getting a lot of new features and we might have those plots available soon (we already have them for the rate loops).

hamishwillee, 2021-01-27:
@bresch I am thinking of migrating some issues across to the user guide, including this one. Does it make sense still to do so?

bresch, 2021-01-27:
@hamishwillee Yes, you could create an issue "Add position/velocity PID tuning guide" to not forget about it.

hamishwillee, 2021-01-28:
Discussion moved to https://github.com/PX4/PX4-user_guide/issues/1017

closed 2021-01-28 by hamishwillee