PX4/PX4-Devguide #809: RTK Base Position

albert21martins, 2019-05-30:
Hi, I thing it is missing the Documentation part on how to save a current RTK position and reloaded for another day. RTK base station is not moving.

hamishwillee, 2019-05-30:
RTK setup is covered here: https://docs.px4.io/master/en/gps_compass/rtk_gps.html
Does it include the information you need?

hamishwillee, 2019-05-30:
FYI, as a general rule the User guide is setup for existing frames, while devguide is the information you need to add new frames and modify core behaviour. There is some overlap, and over time I expect we might merge the two.

albert21martins, 2019-06-05:
You mean, this question should rather be in userguide and not in the developer guide? If so, I am sorry I mix it up. About RTK, there is a option in QGroundControl in order to savee Current Base Station, but I do not know how to get the data of the base station, as it seems, it has to be manually introduced. Is this functionallity operative? 
[image]

hamishwillee, 2019-06-06:
That sounds like a QGC or support question. 

@DonLakeFlyer How do the new RTK settings work/how do you get initial position if you want to enter the settings manually. How does saving/restoring work?

What I would **guess** happens here is that you run **Perform Survey in** to get current values, which would then _seed_ the *Use Specified Base Position** option. You could then save the values. When you next load QGC you could just reuse them, provided you didn't move the base station.

Presumably if you toggle **Perform Survey in** radio button and back the values that are stored remain stored. However if you run the survey in again the saved values would revert to whatever Survey in comes up with.

Is that how it works?

DonLakeFlyer, 2019-06-06:
Once the survey in reports as "valid" the "Save Current Base Position" should enable which will then allow you to copy in the current values. I'm not quite sure exactly when that happens. @bkueng would know more.

DonLakeFlyer, 2019-06-06:
In case not clear: As @hamishwillee says in order to make the survey in  values available you need to start in "Perform Survey-In" mode. Then you can copy them with the button and switch to "Use Specified Base Position"

albert21martins, 2019-06-08:
I tried it on friday and it worked, so thank to everyone for the explanation. I just do not understand why there is a min survey accuracy of 0.5m as RTK is supposed to achieve better accuracy. Am I wrong?

DonLakeFlyer, 2019-06-09:
> I just do not understand why there is a min survey accuracy of 0.5m as RTK is supposed to achieve better accuracy.

@bkueng Any comments on that?

hamishwillee, 2019-06-11:
> In case not clear: As @hamishwillee says in order to make the survey in values available you need to start in "Perform Survey-In" mode. Then you can copy them with the button and switch to "Use Specified Base Position"

1. The button is part of "Use Specified Base Position" group, so to be technical you have to switch to "Use Specified Base Position" before you can press the button.
2. The button is enabled after you have run RTK survey in - does it STAY enabled if you change the values in the "Use Specified ..." section. Ie are there values persistent until next survey, or until you restart QGC, or until a field is changed?
3. The name of the button is a little unhelpful "Save Current Base Position" because this could reflect whatever is in the fields (not RTK survey results). How about "Copy Last Survey-In Position" or "Copy Survey-In Position"

bkueng, 2019-06-11:
> I just do not understand why there is a min survey accuracy of 0.5m as RTK is supposed to achieve better accuracy.

You can set it as low as you want, the survey-in will just take longer. This defines the global accuracy.

albert21martins, 2019-06-12:
I just downloaded QGroundControl and I am not allow to set it less than 0.5m:
[image]
Any hints?

bkueng, 2019-06-13:
You can safely go as low as you want. @DonLakeFlyer can you reduce the minimum to 0.01?

hamishwillee, 2019-06-13:
As an aside ... "The maximum accuracy allowed prior to completing Survey-In" - is "maximum" correct here, given that the value decreases as accuracy improves? Should it be: "The minimum accuracy value before  Survey-In can complete"?

DonLakeFlyer, 2019-06-13:
> can you reduce the minimum to 0.01?

@bkueng Sure but then will someone just say they have a new-fangled foobar that can go below that. Wonder if I should change to something like 0.0000001?

DonLakeFlyer, 2019-06-13:
@albert21martins For a workaround till QGC is changed you can just use Force Save with the value you want.

hamishwillee, 2019-06-14:
@DonLakeFlyer And further to https://github.com/PX4/Devguide/issues/809#issuecomment-501557307 and Albert's question - is there actually any risk to changing the value in this case?

bkueng, 2019-06-14:
> @bkueng Sure but then will someone just say they have a new-fangled foobar that can go below that. Wonder if I should change to something like 0.0000001? It seems like it should really allow anything > 0, but QGC meta data can't support that.

You can use 1mm, but if you really want to go to a global cm/mm level, you're better off using an RTK network service (or some post-processing), because with survey-in alone you cannot reach such accuracies (to my knowledge).

> As an aside ... "The maximum accuracy allowed prior to completing Survey-In" - is "maximum" correct here, given that the value decreases as accuracy improves? Should it be: "The minimum accuracy value that Survey-In must achieve before it can complete"?

I'd say both ways work.

DonLakeFlyer, 2019-06-14:
Thanks for the info.

I'll change to 0.01. The only risk to setting the value too low is that survey-in will never be achieved. I'll also change to "The minimum accuracy value that Survey-In must achieve before it can complete." I think that's slightly better.

closed 2019-06-14 by DonLakeFlyer

hamishwillee, 2019-06-16:
> ... The only risk to setting the value too low is that survey-in will never be achieved. 

That probably is a bit of a risk, particularly if the market ends up having systems with minimum values that are quite different. When someone does survey-in is it obvious what condition completion is waiting on?