PX4/PX4-Devguide #816: Offboard nonexistent coordinate frame

tuloski, 2019-06-17:
In the offboard documentation: https://docs.px4.io/en/flight_modes/offboard.html
it says that the coordinate frame "MAV_FRAME_BODY_NED" is supported, but it doesn't exists in MAVlink coordinate frames list and actually it doesn't make any sense (what is NED in body?).

hamishwillee, 2019-06-17:
Yes it does exist in the frames list - the link on offboard.html is clickable and goes to: https://mavlink.io/en/messages/common.html#MAV_FRAME_BODY_NED

But ... when you click the link you will find that the description is indeed a little unhelpful. 
Essentially in this frame 
- Positions are relative to the EKF Origin in NED frame - ie north, east, down from wherever EKF thinks "home" is.
- Velocities are relative to the current vehicle heading. So you can use this to specify speed forward, right, down. 

The BODY bit just refers to the fact that velocities follow vehicle rather than NED.

closed 2019-06-17 by hamishwillee

tuloski, 2019-06-17:
Oh sorry, I didn't see that.

BTW I'm still confused. A FRAME should be a SINGLE coordinate frame IMO. Based on your description it seems that the pose is in NED and the velocities in BODY. So there are two reference frames. And the pose is not "coordinated" with the velocity. It's weird I must say.
Also what is z velocity positive? Down? Up? It should be specified (in MAVlink documentation) that body in this case is either FRD or FLU.
But I'd avoid at all to use this. Either use full ned (MAV_FRAME_LOCAL_NED) or full body (MAV_FRAME_BODY_FRD).

tuloski, 2019-06-17:
Oh wait maybe I see the point. It's like a BODY without roll and pitch.

hamishwillee, 2019-06-27:
Like I said, the definitions are confusing. I've never understood _why_ they have to be, just what they are.

PS Sorry for delay - was at the dev summit and pretty occupied most of the time.