PX4/PX4-Devguide #820: Doc Bug: Arch Linux · PX4 v1.9.0 Developer Guide

yuhaim, 2019-06-23:
As described in https://github.com/PX4/Firmware/issues/11160
Should add 'ant' installation in this stage

# Common dependencies for all targets
sudo pacman -Sy --noconfirm \
    base-devel make cmake ccache git \
    ninja python-pip tar unzip zip vim wget

Bug Page: [Arch Linux · PX4 v1.9.0 Developer Guide](https://dev.px4.io/en/setup/dev_env_linux_arch.html)

hamishwillee, 2019-06-27:
So is this just a matter of appending "ant" after wget? - ie
```
# Common dependencies for all targets
sudo pacman -Sy --noconfirm \
    base-devel make cmake ccache git \
    ninja python-pip tar unzip zip vim wget
```

yuhaim, 2019-06-27:
Yes, I fixed this issue in manjaro by installing 'ant'

PR #824 (2019-07-01): Archlinux - add ant as dependency
Fixes #820
closed 2019-07-01 by julianoes