PX4/PX4-Devguide #830: ubuntu_sim_ros_melodic.sh fails on Ubuntu 18.04

julianoes, 2019-07-05:
I just ran [ubuntu_sim_ros_melodic.sh](https://raw.githubusercontent.com/PX4/Devguide/v1.9.0/build_scripts/ubuntu_sim_ros_melodic.sh) and it failed as follows:

```
E: Unable to locate package python-catkin-tools
ubuntu_sim_ros_melodic.sh: line 60: catkin: command not found
Writing /home/julianoes/catkin_ws/src/.rosinstall

update complete.
- git:
    local-name: mavlink
    uri: https://github.com/mavlink/mavlink-gbp-release.git
    version: release/kinetic/mavlink/2019.6.7-1

usage: rosinstall_generator [-h] [--debug] [--verbose] --rosdistro ROSDISTRO
                            [--from-path [FROM_PATH [FROM_PATH ...]]]
                            [--repos [reponame [reponame ...]]]
                            [--upstream | --upstream-development]
                            [--deps | --deps-up-to [DEPS_UP_TO [DEPS_UP_TO ...]]]
                            [--deps-depth N] [--deps-only]
                            [--wet-only | --dry-only | --catkin-only | --non-catkin-only]
                            [--exclude [EXCLUDE [EXCLUDE ...]]]
                            [--exclude-path [EXCLUDE_PATH [EXCLUDE_PATH ...]]]
                            [--flat] [--tar]
                            [pkgname [pkgname ...]]
rosinstall_generator: error: argument --rosdistro is required
     Performing actions: 

     Add new elements:
  mavlink

Config changed, maybe you need run wstool update to update SCM entries.
Overwriting /home/julianoes/catkin_ws/src/.rosinstall

update complete.
[mavlink] Fetching https://github.com/mavlink/mavlink-gbp-release.git (version release/kinetic/mavlink/2019.6.7-1) to /home/julianoes/catkin_ws/src/mavlink
Cloning into '/home/julianoes/catkin_ws/src/mavlink'...
remote: Enumerating objects: 835, done.
remote: Counting objects: 100% (835/835), done.
remote: Compressing objects: 100% (627/627), done.
remote: Total 25473 (delta 258), reused 732 (delta 187), pack-reused 24638
Receiving objects: 100% (25473/25473), 7.48 MiB | 7.79 MiB/s, done.
Resolving deltas: 100% (10943/10943), done.
[mavlink] Done.
ubuntu_sim_ros_melodic.sh: line 75: rosdep: command not found
ubuntu_sim_ros_melodic.sh: line 78: rosdep: command not found
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package geographiclib
Downloading dependent script 'install_geographiclib_datasets.sh'
--2019-07-05 08:27:42--  https://raw.githubusercontent.com/mavlink/mavros/master/mavros/scripts/install_geographiclib_datasets.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.240.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.240.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1018 [text/plain]
Saving to: ‘STDOUT’

-                                                                   100%[=================================================================================================================================================================>]    1018  --.-KB/s    in 0s      

2019-07-05 08:27:42 (11.3 MB/s) - written to stdout [1018/1018]

bash: line 26: hash: geographiclib-get-geoids: not found
bash: line 30: hash: geographiclib-datasets-download: not found
OS not supported! Check GeographicLib page for supported OS and lib versions.
ubuntu_sim_ros_melodic.sh: line 99: catkin: command not found
ubuntu_sim_ros_melodic.sh: line 104: /home/julianoes/catkin_ws/devel/setup.bash: No such file or directory
```

What happened to `python-catkin-tools`?

FYI: @TSC21 @mrivi

TSC21, 2019-07-05:
What OS are you running?

julianoes, 2019-07-05:
Ubuntu 18.04 but it looks like this went wrong even earlier in the script:
```
Executing: /tmp/apt-key-gpghome.2z8lR8b7XL/gpg.1.sh --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116
gpg: key 5523BAEEB01FA116: public key "ROS Builder <rosbuild@ros.org>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Hit:1 http://ch.archive.ubuntu.com/ubuntu bionic InRelease
Hit:2 http://ch.archive.ubuntu.com/ubuntu bionic-updates InRelease                             
Hit:3 http://ch.archive.ubuntu.com/ubuntu bionic-backports InRelease                                                                                          
Hit:4 http://packages.osrfoundation.org/gazebo/ubuntu-stable bionic InRelease                                                                                 
Get:5 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]                                      
Get:6 http://packages.ros.org/ros/ubuntu bionic InRelease [4'669 B]              
Err:6 http://packages.ros.org/ros/ubuntu bionic InRelease                        
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
Reading package lists... Done                                  
W: GPG error: http://packages.ros.org/ros/ubuntu bionic InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
E: The repository 'http://packages.ros.org/ros/ubuntu bionic InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-melodic-desktop-full
sudo: rosdep: command not found
ubuntu_sim_ros_melodic.sh: line 41: rosdep: command not found
ubuntu_sim_ros_melodic.sh: line 46: /opt/ros/melodic/setup.bash: No such file or directory
```

julianoes, 2019-07-05:
It doesn't actually add the sources for `melodic`:
```
cat /etc/apt/sources.list.d/ros-latest.list 
deb http://packages.ros.org/ros/ubuntu bionic main
```

mrivi, 2019-07-05:
@julianoes I think this is the fix needed https://github.com/PX4/containers/pull/190

TSC21, 2019-07-05:
The keys were updated here: https://github.com/PX4/Devguide/pull/826.

mrivi, 2019-07-05:
From @julianoes output 
> Executing: /tmp/apt-key-gpghome.2z8lR8b7XL/gpg.1.sh --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116

this is still the old key

TSC21, 2019-07-05:
@julianoes as far as I can tell you are not using the latest script. Use the master one instead of 1.9.0. @hamishwillee we probably need to hotfix it in 1.9.0 as well.

julianoes, 2019-07-05:
I think that's it, thanks @mrivi. I guess the fact that melodic can be installed from the `bionic` deb was confusing me but is actually fine.

So the problem is that for v1.9.0 the key is still old.

TSC21, 2019-07-05:
> I think that's it, thanks @mrivi. I guess the fact that melodic can be installed from the `bionic` deb was confusing me but is actually fine.
> 
> So the problem is that for v1.9.0 the key is still old.

Correct.

julianoes, 2019-07-05:
Thanks for the help @mrivi and @TSC21!

PR #831 (2019-07-05): [BACKPORT] ROS install script update (#826)
We need to update the new GPG key for v1.9 as well.

Fixes #830.
closed 2019-10-30 by hamishwillee