PX4/PX4-Devguide #836: Unable to find documentation on how to use JTAG for debugging Pixhawk 4

echoGee, 2019-07-14:
Some of the pages that existed before is not available 

https://pixhawk.org/dev/jtag/black_magic_probe
https://pixhawk.org/dev/jtag/start?s[]=arm&s[]=none&s[]=eabi&s[]=gdb
https://pixhawk.org/dev/gdb_cheatsheet?s[]=arm&s[]=none&s[]=eabi&s[]=gdb

hamishwillee, 2019-07-14:
This is true. Though those docs were pretty much useless/out of date.

The story now is that you're better off using the Dronecode probe (a.k.a. Pixhawk probe) which is more or less the same as a black magic probe but with the dronecode connector: https://shop.titaneliteinc.com/index.php?route=product/product&product_id=1294

See 
* https://dev.px4.io/v1.9.0/en/debug/system_console.html#connecting-via-dronecode-probe
* https://youtu.be/KZkAM_PVOi0?t=828
* https://dev.px4.io/v1.9.0/en/software_update/stm32_bootloader.html#black-magic--dronecode-probe

@julianoes I want to create a page on debug probes that all these other docs can point do. Do you think you we might be able to had a chat about this?

echoGee, 2019-07-14:
1. Is dronecode probe a mix of JTAG for STM32 + an FTDI equivalent ? 
2. This page (https://dev.px4.io/v1.9.0/en/debug/system_console.html#connecting-via-dronecode-probe) mentions that its a serial interface. But it does JTAG too ? 
3. The above page does not mention connection info for Pixhawk 4 boards. Any documentation on how it connects to this board ? 
4. Is eclipse or qt the preferred IDE for pixhawk now ? The video (https://youtu.be/KZkAM_PVOi0?t=828 ) shows eclipse, but most documentation says eclipse isn't the officially supported. 
5. If any of the GUI/JTAG users record a 30 second video of how the IDE is being used, I think it would be very helpful. It could be stopgap till someone writes a good documentation page.

hamishwillee, 2019-07-15:
Hi @echoGee 

> The above page does not mention connection info for Pixhawk 4 boards. Any documentation on how it connects to this board ?

http://docs.px4.io/master/en/flight_controller/pixhawk4.html#debug-port

> Is eclipse or qt the preferred IDE for pixhawk now ? The video (https://youtu.be/KZkAM_PVOi0?t=828 ) shows eclipse, but most documentation says eclipse isn't the officially supported.

Neither. The IDE that we are moving to is VS Code. It isn't yet fully supported (only works on Linux) but it has the most "potential" and I know that Julian is using it for hardware debugging.

@julian, can you help with 1, 2, 5. I'd love that video too, because it would make documenting much easier!

echoGee, 2019-07-17:
> Hi @echoGee
>
>
> http://docs.px4.io/master/en/flight_controller/pixhawk4.html#debug-port
>
>
> Neither. The IDE that we are moving to is VS Code. It isn't yet fully supported (only works on Linux) but it [...]
@julianoes : Yes, this would help :)

julianoes, 2019-07-18:
Thanks for the note, yes I'm aware this needs improvement. To me this should all be in the dev guide and not in the user docs, right @hamishwillee?

hamishwillee, 2019-07-18:
@julianoes Yes, it should be in the devguide. 

I have a PR for vscode in place, but I personally can't get SITL to run with it, so haven't released it. Once that is sorted was hoping to add info on using it with hardware debugging tools. 

And hoping to create a page that pulls in all those examples on using various hardware probes into one place that the other docs link to.

echoGee, 2019-09-11:
@hamishwillee @julianoes : Could you kindly give an update on this ?

hamishwillee, 2019-09-11:
No progress my side. @julianoes would be good to get a short video showing how to do this on vscode? I'm tempted to put my PR for the docs in and then iterate them via real world testing.

Note, there is another eclipse based PR for J-Link debugging here https://github.com/PX4/Devguide/pull/864 - but it is still waiting on some clarifications like "how do I get the Pixhawk debug adaptor".

hamishwillee, 2019-11-13:
@julianoes 
- There is now a topic on this for Eclipse/Jtag: http://dev.px4.io/master/en/debug/eclipse_jlink.html
- Vscode docs are in too: http://dev.px4.io/master/en/setup/vscode.html

What do we need to know in order to create a VSCode hardware debugging topic?

Note, the problem with these docs is that they don't describe the physical connection to the debug port. Do you know how to do this for recent boards if you DONT have a Pixhawk Debug Adaptor board?

julianoes, 2019-11-14:
I'm going to tag @dinomani to help here.

dagar, 2019-11-14:
Something simple describing the debug connector (similar to http://dev.px4.io/master/en/debug/system_console.html) would be helpful. 

https://pixhawk.org/pixhawk-connector-standard/

[image]

Unfortunately there's quite a bit of variation between boards.

hamishwillee, 2019-11-15:
@dagar I started the process in https://github.com/PX4/Devguide/pull/893 with help from DavidS. There are a bunch of open questions - the most important ones being whether fmu 2 and 3 have a debug port, and what the connector and pinout is.