PX4/PX4-Devguide #839: No documentation on how px4 main processor and IO processor works

echoGee, 2019-07-17:
An outline like this image, for example, seems to be an important information to have for the split up of tasks. 
[image]

This doesn't seem to be in the docs

hamishwillee, 2019-07-18:
@echoGee Yes it does. You will find that the hardware section on the devguide is fairly "thin": http://dev.px4.io/master/en/hardware/

This should not be unexpected - the people who have the expertise to write it or use it is a fairly small number. Which is no to say that we don't want it or welcome it.

Where would you put this? What else would you include?

@david_s5 @dagar @bkueng Is this correct, complete, sufficient? Do you have an opinion on where this could be placed to be most useful. My thinking is possibly somewhere in http://dev.px4.io/master/en/debug/reference-design.html

bkueng, 2019-07-18:
Yes this is helpful and can go under http://dev.px4.io/master/en/debug/reference-design.html (and mention which HW has an IO). I'd add that the link between the 2 is an UART.

PR #841 (2019-07-22): Add diagram showing main/io functions
Fixes #839

@bkueng @echoGee This look OK?

Note:
1. I added UART to the diagram
2. Rather than list what does and does not have the board, I noted the typical naming convention. While not fixed in stone, I actually think that is a little more future proof.

Questions:
1. @echoGee - do you have the source for that image? Could I have that so if things modify in future we can update
2. IF safety switch and LED go through the IO does this mean that these are not present if IO is removed?
closed 2019-07-22 by bkueng