PX4/PX4-Devguide #842: document render problem in https://dev.px4.io/

TianmingLi, 2019-07-22:
in https://dev.px4.io/v1.9.0/en/airframes/adding_a_new_frame.html

the note render result is like this
```
In short, the output of this mixer would be SERVO = ( (roll input -0.6 + 0) + (pitch input 0.65 + 0) ) * 1 + 0
```

which is confusing

---

the github https://github.com/PX4/Devguide/blob/master/en/airframes/adding_a_new_frame.md

markdown render result is 
```
In short, the output of this mixer would be SERVO = ( (roll input * -0.6 + 0) + (pitch input * 0.65 + 0) ) * 1 + 0
```

correct

hamishwillee, 2019-07-23:
Thanks! Actually both versions have identical source, and should be rendering badly. This is because the * used to show multiplication is used as markup for "bold". I have created a fix, which is under review. Note that normally I won't fix things older versions - just master.

hamishwillee, 2019-07-25:
All done (should appear live soon). Note that I did fix v1.9.0 too.

PR #844 (2019-07-25): Add new frame - fix mixer note rendering
Fixes #842

- [ ] Add to v1.9 docs when this is merged.

@bkueng This was rendering badly because the `*` were being interpretted as bold markup. First change was to escape all of those.
[image]

What I also did though is change the first section from `(roll input \* -0.6 + 0) ` to `(roll input \* -0.6 + 0)  \* 1` (ie add `\* 1` at the end) to reflect the scaling - this matches the second part of the markup.

At the very end there is a `+0` (see below). Is this to reflect the "0" line, or something else?
```
> **Note** In short, the output of this mixer [... cut]
closed 2019-07-25 by hamishwillee