PX4/PX4-Devguide #843: Offboard mode with RC requires some description IMO

tuloski, 2019-07-22:
I didn't find much documentation about going in OFFBOARD mode with an RC switch.
Here it is not mentioned at all: https://docs.px4.io/v1.9.0/en/flight_modes/offboard.html.

Moreover I'm having some troubles with it, and without documentation I don't know if it is expected or not. For example if entering OFFBOARD via mavlink message, it is impossible to turn off the OFFBOARD with the RC switch which I'd expect.

hamishwillee, 2019-07-23:
H @tuloski ,

Sorry to hear you're having problems. The documentation is pretty sufficient in that there should be no implications of switching out of offboard mode via an RC switch - and that is what is said in the line "This mode is automatic (RC control is disabled by default except to change modes).". 

There are some implications of switching into the mode, in that you will only be able to switch in if your stream off offboard messages is ok (and it says that in "The vehicle must be already be receiving a stream of target setpoints before this mode can be engaged.").

> For example if entering OFFBOARD via mavlink message, it is impossible to turn off the OFFBOARD with the RC switch which I'd expect.

You expect correctly. You should be able to switch in and out of offboard mode using an RC switch whether you entered it using a mavlink message or via a switch. 

Can you verify that you can switch between other modes - ie from position to mission or similar? My guess is that either the mode change message is not being sent by your switch OR it is trying to switch into a mode that it not valid for some reason. For example, you won't be able to switch into position mode if you have no GPS. 

Have you got a flight log uploaded to Flight review that shows the problematic behaviour?

tuloski, 2019-07-23:
Hi, thanks for the reply. I try to explain better, and today I did more testing.

About this>
> "This mode is automatic (RC control is disabled by default except to change modes).".

Also this is not clear to me. It says that with "COM_RC_OVERRIDE" enabled the user can "override" AUTO modes. But it is not clear to me what overriding means. I enabled it and when in offboard mode I can move any stick but nothing happens. Does "override" means that the mode should revert back to the previous mode or that the pilot can control the movements of the drone? Sorry, but as a (maybe dumb) user I'm not understanding this.

> You expect correctly. You should be able to switch in and out of offboard mode using an RC switch whether you entered it using a mavlink message or via a switch.

I did much more testing and this is what is happening. If I use the switch I have no problems at all. For example the switch > 1750 enables OFFBOARD and < 1750 disables it. Now I have a 3 position switch which are around 1000, 1500 and 2000.  If I have the switch at 1500 and I enter OFFBOARD with mavlink message and then move the switch to 1000 it doesn't exit OFFBOARD, but I have to "enable" the offboard moving the switch at 2000 and then back to 1000. It's a bit weird for me but I can understand it because it is supposed to be a switch with 2 positions.

hamishwillee, 2019-07-23:
>> "This mode is automatic (RC control is disabled by default except to change modes).".

> Also this is not clear to me. It says that with "COM_RC_OVERRIDE" enabled the user can "override" AUTO modes. But it is not clear to me what overriding means. 

What this means is that normally when you're in an automatic mode (e.g. missions, landing, offboard) when you move the sticks nothing happens. If you have `COM_RC_OVERRIDE` enabled then moving the sticks immediately puts the vehicle into position mode.

If `COM_RC_OVERRIDE` is not enabled then moving the sticks does nothing, but you should be able to switch modes still. 

You're not being dumb. I had to look it up - and the behaviour literally just changed this week so that COM_RC_OVERRIDE is enabled by default on master on multicopters and VTOLS in multicopter mode. 

> I did much more testing and this is what is happening. If I use the switch I have no problems at all. For example the switch > 1750 enables OFFBOARD and < 1750 disables it. Now I have a 3 position [...]

Moving a switch can only enable a mode - not explicitly disable it. So if you move out of offboard switch position, but you don't enable some other mode the system doesn't send a new mode change. Could that explain your behaviour?

tuloski, 2019-07-23:
> What this means is that normally when you're in an automatic mode (e.g. missions, landing, offboard) when you move the sticks nothing happens. If you have `COM_RC_OVERRIDE` enabled then moving the [...]

This is not happening to me. I have it enabled (not by default) and when in OFFBOARD mode I move the sticks, it doesn't return in position, nor altitude nor manual. It stays in offboard.

> Moving a switch can only enable a mode - not explicitly disable it. So if you move out of offboard switch position, but you don't enable some other mode the system doesn't send a new mode change. Could that explain your behaviour?

Yeah this makes sense.

hamishwillee, 2019-07-23:
@MaEtUgR Please see this. When COM_RC_OVERRIDE is enabled it should switch out of offboard right?

mhkabir, 2019-07-23:
I believe it only works for Auto modes.

hamishwillee, 2019-07-24:
Thanks @mhkabir ! If that is the case, then all the behaviour is explained and this can be closed. @tuloski As I understand it you just need to set up the mode switching properly on your RC.

closed 2019-07-24 by hamishwillee

tuloski, 2019-07-24:
Yes, but the reference of COM_RC_OVERRIDE and "This mode is automatic" should then be removed from the docs about OFFBOARD.

hamishwillee, 2019-07-25:
> Yes, but the reference of COM_RC_OVERRIDE and "This mode is automatic" should then be removed from the docs about OFFBOARD.

Yes, I fixed that, in master, in #535