PX4/PX4-Devguide #847: Document new PID implementation for rate and velocity controllers

hamishwillee, 2019-07-31:
New PID implementation described in https://github.com/PX4/Firmware/issues/12362 and added in  https://github.com/PX4/Firmware/pull/12472

Document what the PID looks like in relevant documents - e.g. http://dev.px4.io/master/en/flight_stack/controller_diagrams.html?q=

Document impact on tuning - e.g. https://docs.px4.io/master/en/config_mc/pid_tuning_guide_multicopter.html and https://docs.px4.io/master/en/config_fw/pid_tuning_guide_fixedwing.html

@bresch Can you take first shot at this? The tuning guides are IMO particularly important.

hamishwillee, 2019-08-21:
Howdy @bresch, I know this is in your roadmap. Do you have an ETA?

hamishwillee, 2019-09-02:
@bresch I want it all, and I want it now :-)

bresch, 2019-09-03:
fixed by https://github.com/PX4/px4_user_guide/pull/570

closed 2020-04-07 by hamishwillee