PX4/PX4-Devguide #848: gstreamer not working

hardikrshah007, 2019-08-06:
I tried installing Gstreamer using the below commands but it still doesn't work me. I want to get video stream from Gazebo inside QGroundControl .

list=$(apt-cache --names-only search ^gstreamer1.0-* | awk '{ print $1 }' | grep -v gstreamer1.0-hybris)
-> sudo apt-get install $list

I tried this step and it doesnt work for me on Ubuntu 18.04. Below is the error
The following packages have unmet dependencies:
gstreamer1.0-python3-dbg-plugin-loader : Conflicts: gstreamer1.0-python3-plugin-loader but 1.14.5-0ubuntu118.04.1 is to be installed
gstreamer1.0-python3-plugin-loader : Conflicts: gstreamer1.0-python3-dbg-plugin-loader but 1.14.5-0ubuntu118.04.1 is to be installed
E: Unable to correct problems, you have held broken packages.

Please if someone has found a solution or tried on Ubuntu 18.04

hamishwillee, 2019-08-07:
Can you confirm how you installed Gazebo (ie which instructions you followed), what version of PX4.

Did you follow these instructions to set up video?: https://dev.px4.io/master/en/simulation/gazebo.html#video-streaming

hardikrshah007, 2019-08-07:
I installed ROS-melodic on Ubuntu 18.04 ( so Gazebo 9 is installed by default). 
For PX4, I am using the latest build from github.  Yes, I followed the instructions in the link you mentioned and everything else works except the camera plugin.

hamishwillee, 2019-08-08:
@lbegani Can you advise on this one?

lbegani, 2019-08-09:
> @lbegani Can you advise on this one?

I have not installed gstreamer on Ubuntu18.04 (yet). Instead of trying to install all the gstreamer packages using the automated command, you can selectively install what is required for video streaming use case. For example - 
```
$ sudo apt-get install gstreamer-1.0
$ sudo apt-get install libgstreamer-plugins-base1.0-dev
$ sudo apt-get install libgstrtspserver-1.0-dev  
```
There will be few more, I do not have the full list handy.

hardikrshah007, 2019-08-09:
@lbegani @hamishwillee 
Selectively installing gstreamer worked for me. However, the next issue I am facing is that there is no 
libgazebo_video_stream_widget plugin in gazebo 9 and hence, the video fails again.