PX4/PX4-Devguide #855: uORB Messaging - Docs on new interface

hamishwillee, 2019-08-19:
As discussed [here](https://discuss.px4.io/t/px4-dev-call-august-14-2019/12431/3) there is a new uORB interface. Should result in docs update to [uORB Messaging · PX4 Developer Guide](http://dev.px4.io/master/en/middleware/uorb.html). Assigning to @dagar to at least kick it off.

@bkueng Do you know of any other places that need to be updated?
Not knowing anything about the changes, the things that spring to mind are [First module doc](http://dev.px4.io/master/en/apps/hello_sky.html) and the [Module template](http://dev.px4.io/master/en/apps/module_template.html). If needed, could you sort out those changes?

dagar, 2019-08-19:
It seems we don't actually have much specific documentation in the first place.

For one thing the troubleshooting and common pitfalls no longer apply.
http://dev.px4.io/master/en/middleware/uorb.html#troubleshooting-and-common-pitfalls

Updating the module template and an example might be the next step.

hamishwillee, 2019-08-19:
Thanks. I deleted the troubleshooting/common pitfalls section.

@bkueng Can you update the module template/example?