PX4/PX4-Devguide #875: Starting Gazebo and PX4 Separately

TheLegendaryJedi, 2019-10-07:
Hi @hamishwillee, In [dev guide](http://dev.px4.io/v1.9.0/en/simulation/gazebo.html), starting gazebo separately is not working. For what I read `gazebo_none_ide` don't exist anymore.

I think it's better to hide this documentation part or give an alternative. In my case I wanted to start gazebo with a different world and then start PX4 separately, so ended editing the iris.world (default) to achieve this.

Thanks

hamishwillee, 2019-10-07:
@TheLegendaryJedi Yes, it is a duplicate of an older issue: https://github.com/PX4/Devguide/issues/454

I believe the correct approach is outlined here: https://github.com/PX4/Devguide/issues/454#issuecomment-363729886 but no one has confirmed. If you can confirm this approach makes sense I can update the docs.

cschindlbeck, 2020-08-19:
`gazebo_none_ide` has been replaced with `gazebo___ide`

hamishwillee, 2020-08-20:
@cschindlbeck Thanks.
So in https://github.com/PX4/Devguide/issues/454#issuecomment-363729886 if I just replace gazebo_none_ide with gazebo___ide it will just work? Have you tested this?

hamishwillee, 2020-09-15:
Hi @cschindlbeck Did you see my comment above?

cschindlbeck, 2020-09-15:
Yes, so for me it worked, but since I am a novice px4 user, someone should double check this.