PX4/PX4-Devguide #894: Fail to Build DevGuide Doc

BrightSoulXYHY, 2019-11-18:
Hello I try to clone it and using gitbook to build for static html, but some errors occour
First I try to `gitbook build` it show the module `weak` is requird
```
$ gitbook build
info: parsing multilingual book, with 2 languages
info: 22 plugins are installed
info: 21 explicitly listed
info: loading plugin "youtube"... OK
info: loading plugin "richquotes"... OK
info: loading plugin "collapsible-menu"... OK
info: loading plugin "anchors"... OK
info: loading plugin "page-toc-button"... OK
info: loading plugin "ga"... OK
info: loading plugin "language-picker"... OK
info: loading plugin "custom-favicon"... OK
info: loading plugin "bulk-redirect"... OK
info: loading plugin "toolbar"... OK
info: loading plugin "validate-links"... OK
info: loading plugin "versions-select"... OK
info: loading plugin "mathjax"... OK
info: loading plugin "mermaid"... OK
info: loading plugin "highlight"... OK
info: loading plugin "search"... OK
info: loading plugin "lunr"... OK
info: loading plugin "sharing"... OK
info: loading plugin "fontsettings"... OK
info: loading plugin "theme-dronecode"... OK
info: loading plugin "theme-default"... OK
info: found 749 asset files
warn: "options" property is deprecated, use config.get(key) instead
info:
info: generating language "en"
info: found 137 pages
info: found 7 asset files
warn: "book" property is deprecated, use "this" directly instead
warn: "this.generator" property is deprecated, use "this.output.name" instead
warn: "navigation" property is deprecated
internal/modules/cjs/loader.js:895
    throw err;
    ^

Error: Cannot find module 'weak'
Require stack:
- /mnt/f/Progroming/_Doc_Src/PX4-Devguide/node_modules/dnode/index.js
- /mnt/f/Progroming/_Doc_Src/PX4-Devguide/node_modules/phantom/phantom.js       - /mnt/f/Progroming/_Doc_Src/PX4-Devguide/node_modules/gitbook-plugin-mermaid/index.js
- /home/brightsoul/.gitbook/versions/3.2.3/lib/plugins/loadPlugin.js
- /home/brightsoul/.gitbook/versions/3.2.3/lib/plugins/loadForBook.js
- /home/brightsoul/.gitbook/versions/3.2.3/lib/plugins/index.js
- /home/brightsoul/.gitbook/versions/3.2.3/lib/api/encodeGlobal.js
- /home/brightsoul/.gitbook/versions/3.2.3/lib/api/index.js
- /home/brightsoul/.gitbook/versions/3.2.3/lib/output/getModifiers.js
- /home/brightsoul/.gitbook/versions/3.2.3/lib/output/json/onPage.js
- /home/brightsoul/.gitbook/versions/3.2.3/lib/output/json/index.js
- /home/brightsoul/.gitbook/versions/3.2.3/lib/output/index.js
- /home/brightsoul/.gitbook/versions/3.2.3/lib/index.js
- /usr/lib/node_modules/gitbook-cli/lib/local.js
- /usr/lib/node_modules/gitbook-cli/lib/index.js
- /usr/lib/node_modules/gitbook-cli/bin/gitbook.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:892:15)     at Function.Module._load (internal/modules/cjs/loader.js:785:27)
    at Module.require (internal/modules/cjs/loader.js:956:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at new D (/mnt/f/Progroming/_Doc_Src/PX4-Devguide/node_modules/dnode/index.js:28:20)
    at module.exports (/mnt/f/Progroming/_Doc_Src/PX4-Devguide/node_modules/dnode/index.js:8:12)
    at /mnt/f/Progroming/_Doc_Src/PX4-Devguide/node_modules/phantom/phantom.js:174:13
    at Server.handler (/mnt/f/Progroming/_Doc_Src/PX4-Devguide/node_modules/shoe/index.js:22:9)
    at Server.emit (events.js:210:5)
    at App.emit (/mnt/f/Progroming/_Doc_Src/PX4-Devguide/node_modules/sockjs/lib/sockjs.js:182:27) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/mnt/f/Progroming/_Doc_Src/PX4-Devguide/node_modules/dnode/index.js',          '/mnt/f/Progroming/_Doc_Src/PX4-Devguide/node_modules/phantom/phantom.js',      '/mnt/f/Progroming/_Doc_Src/PX4-Devguide/node_modules/gitbook-plugin-mermaid/index.js',
[... 57 lines cut ...]
npm WARN PX4-Devguide No repository field.
npm WARN PX4-Devguide No README data
npm WARN PX4-Devguide No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! weak@1.0.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the weak@1.0.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/brightsoul/.npm/_logs/2019-11-18T14_45_53_155Z-debug.log
```
I try many methods but it does not work, Please Help me
and the version for nodejs 
```
$ npm -v
6.12.1
$ node -v
v13.1.0
```

Think you

hamishwillee, 2019-11-18:
I build this on Ubuntu 16.04, and so does the docker build: https://github.com/PX4/containers/blob/master/docker/Dockerfile_docs

The issue is caused by the mermaid plugin. If you just want to do testing  (ie not for publishing) the easiest fix is to disable the mermaid plugin. This stops some sequence diagrams rendering but on 95% of pages you can render and see what is going on.

The change needs to be made here https://github.com/PX4/Devguide/blob/master/book.json#L22 - basically you move the "-" in front of mermaid and remove the one from stub-out ...
```
        "-mermaid",
        "stub-out-blocks@git+https://github.com/hamishwillee/gitbook-plugin-stub-out-blocks.git",
```

closed 2019-11-18 by hamishwillee

BrightSoulXYHY, 2019-11-19:
Thanks a lot ，it dose work

bys1123, 2020-03-24:
I met the same issue……

bys1123, 2020-03-24:
https://github.com/chriswessels/gitbook-plugin-mermaid-gb3
I found a mermaid plugin for gitbook3 but it have a different syntax.

bys1123, 2020-03-24:
https://github.com/miao1007/gitbook-plugin-mermaid-cli
this one looks good

bys1123, 2020-03-24:
not good, it need chrome with X11 desktop

bys1123, 2020-03-24:
reference https://github.com/JozoVilcek/gitbook-plugin-mermaid/issues/21
I use `n 4.2.6`
now everything is ok on 18.04

hamishwillee, 2020-03-24:
FYI, I updated to Ubuntu LTS 18.04 and the build all works with the set of plugins in the devguide.

bys1123, 2020-03-25:
I installed node 12 lts on my ubuntu, maybe that's the reason it's not work.

hamishwillee, 2020-03-25:
Maybe. I used the node that comes when you just do `sudo apt install nodejs`

bys1123, 2020-03-25:
That is node 8.10 I guess, I’ll try later.

bys1123, 2020-03-25:
`gitbook build` progress will stop at
```
debug: generate page "concept/mixing.md" 
debug: calling hook "page:before" 
```
with node 8.10.0

bys1123, 2020-03-25:
I tried a new server with apt installed nodejs, it fixed.

bys1123, 2020-03-25:
Trying on old server, if run as "sudo" will stop at
```
debug: generate page "concept/mixing.md" 
debug: calling hook "page:before" 
phantom stdout: Incompatibile SockJS! Main site uses: "0.3.1.7.ga67f.dirty", the iframe: "0.3.4".

phantom stdout: Incompatibile SockJS! Main site uses: "0.3.1.7.ga67f.dirty", the iframe: "0.3.4".

debug: calling hook "page" 
debug: index page concept/mixing.md 
debug: generate page "concept/pwm_limit.md" 
debug: calling hook "page:before" 
debug: calling hook "page" 
debug: index page concept/pwm_limit.md 
debug: generate page "concept/system_startup.md" 
debug: calling hook "page:before" 
debug: calling hook "page" 
debug: index page concept/system_startup.md 
debug: generate page "simulation/README.md" 
debug: calling hook "page:before" 
debug: calling hook "page" 
debug: index page simulation/README.md 
debug: generate page "simulation/jmavsim.md" 
debug: calling hook "page:before" 
```

bys1123, 2020-03-25:
`n 5.9.1` works fine. I‘ll use this. whatever, same result.