PX4/PX4-Devguide #899: Doc Bug: Building the Code · PX4 v1.9.0 Developer Guide

eddyanm, 2019-11-22:
DESCRIBE PROBLEM WITH DOCS HERE

Bug Page: [Building the Code · PX4 v1.9.0 Developer Guide](https://dev.px4.io/v1.9.0/en/setup/building_px4.html)

In Raspberry Pi 2/3 section, 'export AUTOPILOT_HOST' should go before 'make emlid_navio2_cross'.  As is, make cached an empty value if AUTOPILOT_HOST is not set and will cause 'make emlid_navio2_cross upload' to fail as it can't reach the Pi with an empty string.

At least 3 people including myself ran into this issue while following the guide.

hamishwillee, 2019-11-22:
Thanks. Can you please make a docs PR - removes any doubt about the correct way to do this: https://github.com/PX4/Devguide/edit/master/en%2Fsetup%2Fbuilding_px4.md

eddyanm, 2019-11-22:
Got it. PR submitted.

Eddy

hamishwillee, 2019-11-24:
Hi @eddyanm , I can't see that PR [in the list](https://github.com/PX4/Devguide/pulls). Can you point me to the number?

eddyanm, 2019-11-25:
Apparently, I made edits but didn't submit a pull request.  Here it is. https://github.com/PX4/Devguide/pull/900

hamishwillee, 2019-11-25:
Thanks for that. Merged.

closed 2019-11-25 by hamishwillee