PX4/PX4-Devguide #917: Doc Bug: Ubuntu/Debian Linux · PX4 Developer Guide

wujoho, 2020-01-10:
when running step 2:
_source ubuntu.sh_

got warnings and errors:
[image]
[image]
[image]

my system:
A new installed Ubuntu 16.04.6 LTS in Oracle VM VirtualBox

Bug Page: [Ubuntu/Debian Linux · PX4 Developer Guide](https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html)

hamishwillee, 2020-01-13:
@julianoes I can reproduce the above errors on Ubuntu 16.04 and Ubuntu 18.04. For Ubuntu 18.04 the build still seems to "work" (ie `HEADLESS=1 make posix jmavsim` launches simulator). @wujoho Did the above command work after your installation finished on Ubuntu 16.04?

Julian, any ideas?

wujoho, 2020-01-13:
@hamishwillee No, both commands `make px4_sitl jmavsim` or `HEADLESS=1 make posix jmavsim` failed in my Ubuntu 16.04, as below:
[image]
[image]

julianoes, 2020-01-13:
Thanks for the issue @wujoho. This looks indeed broken. I'll look into it.

closed 2020-01-18 by LorenzMeier