PX4/PX4-Devguide #926: Doc Bug: Building the Code · PX4 Developer Guide - Ubuntu 18.04.3 LTS

wujoho, 2020-01-21:
My system: a new installed Ubuntu 18.04.3 LTS in Oracle VM VirtualBox
After installing Ubuntu, rebooted it, then immediately followed:
https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html
the below terminal commands/steps have performed one after one:
`wget https://raw.githubusercontent.com/PX4/Firmware/master/Tools/setup/ubuntu.sh`
`wget https://raw.githubusercontent.com/PX4/Firmware/master/Tools/setup/requirements.txt`
`source ubuntu.sh` (this step having a small issue, see bottom)
_restarted Ubuntu in VM_
The above steps seems worked.
Then I followed:
https://dev.px4.io/master/en/setup/building_px4.html
the below terminal commands/steps have performed one after one:
`sudo apt install git`
`git clone https://github.com/wujoho/Firmware.git`
`cd Firmware`
`make px4_sitl jmavsim`, but it got stuck at this command and there was no simulator started:
[image]

BTY, when I run `source ubuntu.sh`, it failed to download _gcc-arm-none-eabi-7-2017-q4-major-linux.tar.bz2_ at the first try around 60% progress a few times (I have noticed this at least 3 times both in Ubuntu 16 LTS and 18 LTS), it got stuck for about 20 minutes, then it failed and started retrying, as below:
[image]

Bug Page: [Building the Code · PX4 Developer Guide](https://dev.px4.io/master/en/setup/building_px4.html)

hamishwillee, 2020-01-28:
The illegal reflexive operation is odd, but looks like you have the wrong java: http://dev.px4.io/master/en/simulation/jmavsim.html#an-illegal-reflective-access-operation-has-occured
Was this already installed?

The gcc-arm-none-eabi-7-2017-q4-major-linux.tar.bz2  downloads fine for me. It is a big file. You need this to build for NuttX (hardware). It looks like it succeeded second time.

hamishwillee, 2020-01-28:
@julianoes I ran this on latest Firmware on clean Ubuntu 18.04 in a VM and was able to to run `HEADLESS=1 make posix jmavsim` and run the Jmavsim simulator.

But without HEADLESS=1 I get the error below. The docs update you did recently indicate this is "wrong Java" which it cannot be. Previously I would expect an OpenGL error in this case. Thoughts?
My concern is that this error might happen on real ubuntu installation. 

```
BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
Jan 28, 2020 2:19:23 PM java.util.prefs.FileSystemPreferences$1 run
INFO: Created user preferences directory.
3D [dev] 1.6.0-pre12-daily-experimental daily

WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Inconsistency detected by ld.so: dl-lookup.c: 111: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!
```

julianoes, 2020-01-29:
@hamishwillee What  is your Java version?

```
java -version
```
or:
```
java --version
```

hamishwillee, 2020-01-30:
@julianoes It is wrong, but the point is that I did not explicitly install Java - this was a clean Ubuntu. So the script needs to make sure that the right version is used.

```
ubuntu@ubuntu:~/github/Firmware$ java --version
openjdk 11.0.6 2020-01-14
OpenJDK Runtime Environment (build 11.0.6+10-post-Ubuntu-1ubuntu118.04.1)
OpenJDK 64-Bit Server VM (build 11.0.6+10-post-Ubuntu-1ubuntu118.04.1, mixed mode, sharing)
```

julianoes, 2020-01-30:
Aha, got it. Java 8 is installed but not selected. Best would be to find a way to select Java 8 at build time rather than messing with the java default system wide. I'll check if we can do that.

closed 2020-01-31 by dagar