PX4/PX4-Devguide #943: Missing one step in SITL unit tests documentation

roangel, 2020-02-05:
I think there is a missing step here:
https://github.com/PX4/Devguide/blob/6194b65f72615ffbcb4c8ed07a6b2a07a9a9362d/en/test_and_ci/unit_tests.md

In the section `Writing a SITL Unit Test`, we also need to add the test to `platforms/posix/cmake/sitl_tests.cmake`, and it's not mentioned anywhere.

hamishwillee, 2020-02-05:
@jkflying Can you please comment on this ^^^. 

@roangel FYI only, you can create an issue report or PRs from the top of each docs page. Slightly simpler for you because it populates the link to the affected page.
[image]

roangel, 2020-02-05:
> @jkflying Can you please comment on this ^^^.
> 
> @roangel FYI only, you can create an issue report or PRs from the top of each docs page. Slightly simpler for you because it populates the link to the affected page.
> [image]

Aha, ok, thanks!

jkflying, 2020-02-06:
I've actually never worked with these SITL unit tests. There is the instruction already to add your test to the `systemcmds/tests/CMakeLists.txt`, which I think should be sufficient, but I'm not 100% sure.

hamishwillee, 2020-02-06:
Thanks @jkflying . @julianoes Are you any "more sure"?

julianoes, 2020-02-07:
@MaEtUgR might know.