PX4/PX4-Devguide #944: adoptopenjdk vs. java in Development Environment on Mac guide

lrntzmmrl, 2020-02-06:
Following the installation guide for the dev tools on Mac, the guide gives the following instruction to resolve the Java (and Quartz) dependency:
`brew cask install xquartz java`

Brew, however, gave me the following instruction (for Java):
`brew cask install adoptopenjdk`

If adoptopenjdk is the right package, you might want to update the guide. Furthermore, moving the dependencies up would be helpful (for people like me, who don't read ahead ;) ).

https://dev.px4.io/master/en/setup/dev_env_mac.html

julianoes, 2020-02-06:
For jMAVSim simulation we recommend `adoptopenjdk8` here:
https://dev.px4.io/master/en/simulation/jmavsim.html#macos

However, it would  be nice to have the initial setup correct as well.
I'm first trying to include this in the `px4-sim` tap:
https://github.com/PX4/homebrew-px4/pull/50

hamishwillee, 2020-02-06:
@lrntzmmrl Yes, definitely the main entry points need to be corrected. Looking at the link @julianoes provided this can't be handled in brew automatically. Julian, I've created PR which I think fixes this. Feel free to edit directly if you don't like it. #946

PR #948 (2020-02-10): setup: split jMAVSim/Gazebo brew install
This was my attempt to fix #944.
closed 2020-02-10 by hamishwillee